rusel1989/react-native-bluetooth-serial #1: Calling .enable() on Android never seems to resolve

joewood, 2016-07-13:
Using React-native 0.29 on Android 4.4.2.
I'm seeing an exception on Android when attempting to enable Bluetooth. This doesn't result in an error from the promise. The exception is:
```
E/BluetoothAdapter( 2176): Bluetooth binder is null
E/RequestPermissionActivity( 2176): Error: there's a problem starting Bluetooth
```
The permissions are set-up in the manifest as follows:
```
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
```

rusel1989, 2016-07-18:
Thanks for feedback, I am on it

rusel1989, 2016-08-08:
Hey guys sorry for the slight delay i was on vacation ... i will try to resolve this ASAP

rusel1989, 2016-09-09:
Fixed in 0.1.1 , you can also listen for bluetoothEnabled and bluetoothDisabled events and use disable() to turn off bluetooth

closed 2016-09-09 by rusel1989