rusel1989/react-native-bluetooth-serial #2: Is polling required to receive data?

esutton, 2016-08-05:
I want to connect to a Bluetooth Classic SPP device ( not an Arduino device ).

Use Case:
1. Send command to device.
2. Process response from device when it is received.

Is there an API event or something that can be used to asynchronously handle data reception from the Bluetooth device?

Thanks for sharing your work.  Any advice is appreciated.

rusel1989, 2016-08-08:
Hey,
Both use cases will need some fixes/updates ... I just came back from vacation so i will start working on it ... but you know Pull Requests are welcome

jcteague, 2016-08-24:
I'm needing this functionality too.  I'm checking to see if the product owners want me to contribute to this project to make it work. If they do, do you have  general idea of the work.  Feel free to dm me if that's easier.

rusel1989, 2016-09-09:
@jcteague how does it look ? I currently have some free time so if you are not going to do it, i will start working on it

jcteague, 2016-09-09:
I'm on hold right now. I will know more next week.

rusel1989, 2016-09-09:
@jcteague okay so i will start working on it, i just checked it and it seems that there shoudn't be much work on making it work

jcteague, 2016-09-09:
rgr.  I'll check in with you next week and see if there is anything I can
do to help.

FYI, my use case.  I'm connecting to a bluetooth device connected a
vehicle.  I can send an $AT command that will respond with info about the
vehicle (gps, rpms, etc..)

jcrjaci, 2016-09-16:
@jcteague i need to do that as well. Did you use this library or developed something your own?

jcteague, 2016-09-16:
we've punted on this for now, focusing on other requirements.  We will need
to address it at some point.

arvindpdmn, 2016-11-29:
The BTSerial API is documented here: https://github.com/derektu/react-native-bluetooth-serial
This may help in reading data on app when received from device. I think `setDataAvailableCallback()` may help.

timscott, 2016-12-06:
I'm taking over for @jcteague. Any progress on this?

By the way, I can't find a way to receive data from the connected device through polling or any other means. When I ```write```, the resolved value is ```true``` which I'm guessing means the write succeeded; it's not a response from the device.

I guessing I will need to extend this package to implement at least the [read](https://github.com/don/BluetoothSerial#read) function and perhaps [subscribe](https://github.com/don/BluetoothSerial#subscribe). Is that correct?

timscott, 2016-12-06:
After some digging I have discovered that ```read``` and ```subscribe``` (and many other functions) are already implemented. They are just not documented nor used in the [sample app](https://github.com/rusel1989/react-native-bluetooth-serial/blob/master/BluetoothSerialExample/src/App.js).

In regards to the original question, it would appear that [subscribe](https://github.com/don/BluetoothSerial#subscribe) does the job. However, I have not been able to test that yet.

rusel1989, 2016-12-07:
@timscott yep you're right, i didn't test these, so i don't know if they're working correctly ... I don't have much time lately ...

jlucier, 2017-01-09:
@timscott @rusel1989 
I'm trying to do read/write operations similar to what has been discussed here. Any update on whether or not `read` and `subscribe` are functional?

timscott, 2017-01-10:
@jlucier - I've got read working. Realize that read will read what's written which might not be  everything you want. If you need a request-response type implementation, as I do, you will need to read into a buffer (I just append to a string) in a tight synconous loop until you reach a delimiter.

subscribe does not work for me at all.  It's only ever called in response to a successful connect, returning true.

jlucier, 2017-01-10:
@timscott 
Sounds good. I was expecting to have to buffer the messages like that. Is `read` working on the most recent release (the version installed via npm) or something else?

Searingfang, 2017-01-21:
It took some doing but it does appear that subscribe does work.  You have to call Bluetooth.subscribe(delimiter).then(whateverfunction).  That promise will always just return true but to get the data you need to make a listener with Bluetooth.on('data', dataHandler).

timscott, 2017-01-21:
@Searingfang - Ah ha! That makes sense. Great work! I'll try that ASAP.

storerjeremy, 2017-02-10:
It seems I am also working on something similar @timscott Did you manage to get subscribe working? It would be awesome if someone could update the example app to show how to use subscribe?

jlucier, 2017-02-10:
@storerjeremy It does work. I did what @Searingfang said to do and it worked without issue.

storerjeremy, 2017-02-10:
@jlucier Awesome, thanks for the confirmation it works. I'll give it a shot. Cheers

storerjeremy, 2017-02-19:
@jlucier @Searingfang I tried using BluetoothSerial.subscribe('\n').then(whateverfunction) and BluetoothSerial.on('data', dataHandler) in the example app, however when the subscribe method is called it brings up an error "TypeError: _reactNativeBluetoothSerial2.default.subscribe is not a function". Any idea whats going on?

Searingfang, 2017-02-19:
@storerjeremy Based on the package.json in the example app it uses the version of the bluetooth serial library on github.  This is actually a different version than you would get from npm install.  The version on github doesn't have the subscribe feature for some reason.

So probably all you have to do is change the package.json dependancy for react-native-bluetooth-serial from "file:../" to "^0.1.6".  If that doesn't work make a small app from scratch.  Cards on the table, I wasn't able to get the example working when I first tried.

jorge627, 2017-04-06:
Hi
I want confirm that with subscribe it works, but I have a question, Do you know what is the limit time between data for receive it ?, I mean if my app is paired with a bluetooth device that is constantly sending data, what is the maximum frequency to send data from device to the app and obviously that the app can read each data properly ?, for example if the bluetooth device sends data each 10ms the app is in capacity to read each data properly ?

Thanks in advance

pewh, 2017-04-08:
@Searingfang I'm trying to use `subscribe`, but I don't have any idea how to send command. What I'm trying is: after connect to device, send request 'ATI' like [this](http://stackoverflow.com/questions/37340574/auto-determine-the-bluetooth-device-is-obd-or-not). Thanks

Searingfang, 2017-04-26:
@pewh sending a command uses the `write` command.  What you receive is dependent on the device.  If you are trying to use `subscribe` then you have to know the device is delimiting its messages with something.  For instance mine uses  a newline `\r\n`.  So if you `subscribe('\r\n')` then whenever you get a newline from your device Bluetooth serial will dispatch a data event that you can handle with `on('data', dataHandler)`

babyinthesky, 2017-06-19:
for the latest github version, I get the listener for receiving data working by the following way. Note that you have to replace the delimiter '\r\n' with the one upon your own case.

BluetoothSerial.withDelimiter('\r\n').then((res)=>{
          console.log("delimiter setup",res);
          BluetoothSerial.on('read',(data)=>{
            console.log('read',data);
          })
      })

arrygoo, 2017-10-23:
Any updates on doing this without `setInterval` would be sweet

hungtm84, 2019-09-10:
anyone have subscribe function example ? please share.
tks!

jorge627, 2019-09-10:
@hungtm84 I created an small project with bluetooth using the subscribe function ( it also use redux ) you can check it here, it is still a work in progress but the bluetooth is working, may be it can help you.

ichbinvinh, 2019-09-10:
> for the latest github version, I get the listener for receiving data working by the following way. Note that you have to replace the delimiter '\r\n' with the one upon your own case. [...]

Hi @babyinthesky ,

i placed your code in componentWillMount() but no events was catched when my scanner read barcode.

`BluetoothSerial.withDelimiter('\r\n').then(() => { 
        //     BluetoothSerial.on('read', (data) => { 
        //       Toast.showShortBottom('DATA FROM BLUETOOTH 2:'+JSON.stringify(data))
        //     }); 
        //   })  `

I use this version : "react-native-bluetooth-serial": "^1.0.0-rc1"

Thanks in advance

WhaSukGO, 2020-03-21:
Any update? `BluetoothSerial.withDelmiter(...` doesn't seem to work