rusel1989/react-native-bluetooth-serial #5: BluetoothSerial.connect(device.id) always enter in catch err.

jcrjaci, 2016-09-16:
Hi guys,

I'm follow the provider example.

I'm able to:
- check the bluetooth state
- enable and disable
- list the devices
- get unpair the devices

but when i try to connect to a device through
```
connect (device) {
    this.setState({ connecting: true })
    BluetoothSerial.connect(device.id)
    .then((res) => {
      Toast.showLongBottom(`Connected to device ${device.name}`)
      this.setState({ device, connected: true, connecting: false })
    })
    .catch((err) => Toast.showLongBottom(err))
}
```
That always enter in the catch err.
I saw that the follow error appear in log section:
```
java.io.IOException: read failed, socket might closed or timeout, read ret: -1
```
I am testing this by connecting the app to mobile devices, but in the future i need to access to a obdII device.

![14360449_1381705821857688_1895773428_o]
(https://cloud.githubusercontent.com/assets/3009975/18592688/fd83059a-7c2f-11e6-90c3-c139ac4171ec.png)

after i pair the device, the error ocurrs, but in the bluetooth settings of the phone, the phone appear paired.
[image]

any idea?

Thanks in advance.

rusel1989, 2016-09-19:
Hey, this lib is based on cordova plugin [BluetoothSerial](https://github.com/don/BluetoothSerial) which doesn't support android to android or ios to ios connections. I will add mention about this fact into readme and when i will have some spare time i will check how to implement this functionality. And you know PRs are welcome :-)

rusel1989, 2016-11-30:
Closing for now

closed 2016-11-30 by rusel1989

sun2rise, 2016-12-28:
Any news about this? 
Thanks!