rusel1989/react-native-bluetooth-serial #6: Property 'on'

gillestasse, 2016-09-22:
When I import (after install) :
`import BluetoothSerial from 'react-native-bluetooth-serial'`

I get the following error on the device :
`ExceptionsManager.js:61Cannot set property 'on' of undefined`

rusel1989, 2016-09-22:
@gillestasse Hey, what version of react-native are you using and, did you link ios and android native code ?

gillestasse, 2016-09-22:
Hi.

Sorry for the lack of details.

I linked it on Android.

React-native : 

```
react-native-cli: 1.0.0
react-native: 0.33.0
```

rusel1989, 2016-09-22:
@gillestasse okay that should work without any problem, it seems to me that it is not properly linked ... did you check it is correctly linked in `android/src/main/java/com/appname/MainApplication.java`

... it should look something like this
[image]

gillestasse, 2016-09-22:
@rusel1989 Thanks a lot for you fast answer. It works now, i don't have any error message. Did I miss something in the doc ? I tried to do it through the automatic method (rnpm link). When I ran rnpm link a second time it confirmed that the library was already linked. Is it necessary to do the automatic and the manual method ?

rusel1989, 2016-09-22:
You're welcome. It wasn't your fault, i should update the docs, rnpm is deprecated, it has been moved to react-native core since version 0.27, i will update the docs and close. You can read the more [here](https://github.com/rnpm/rnpm).

closed 2016-09-22 by rusel1989

kranthi1216, 2025-03-15:
how to resolve this issue in iOS