rusel1989/react-native-bluetooth-serial #9: discoverUnpairedDevices() returns empty array

nehasalian, 2016-10-25:
Tried to run  BluetoothSerialExample and i am unable to scan unpaired devices

BluetoothSerial.discoverUnpairedDevices()
      .then((unpairedDevices) => {
        **console.log("discoverUnpaired",unpairedDevices);** **// Returns empty array**
        const devices = this.state.devices
        const deviceIds = devices.map((d) => d.id)
        unpairedDevices.forEach((device) => {
          if (deviceIds.indexOf(device.id) < 0) {
            devices.push(device)
          }
        })
        this.setState({ devices, discovering: false })
      })

**react-native version : 0.32.0
mobile os : android**

wzh19960613, 2016-10-29:
Me too...My react-native version is 0.36 and android is 6.0.1

ErreMalote, 2016-11-24:
discoverUnpaired () {
    if (this.state.discovering) {
      return false
    } else {
      this.setState({ discovering: true })
      debugger;
      BluetoothSerial.discoverUnpairedDevices() 
      .then((unpairedDevices) => { `unpairedDevices = []`
        const devices = this.state.devices `devices = [Object, Object, Object]`
        const deviceIds = devices.map((d) => d.id) `deviceIds = ["B8:69:C2:8F:40:D0", "8C:1A:BF:77:4B:9F", "10:A5:D0:B0:27:BE"]`
        debugger
        unpairedDevices.forEach((device) => {
          if (deviceIds.indexOf(device.id) < 0) {
            devices.push(device)
          }
        })
        this.setState({ devices, discovering: false })
      })
    }
  }
is the **discoverUnpairedDevices()** being worked on?

android sdk 23
react-native 0.37

rusel1989, 2016-11-24:
Hey, I've checked the method today ... it seems like the code should be okay ... could you please specify the devices you are trying to discover ?

ErreMalote, 2016-11-24:
for me it is an Adafruit Bluefruit LE. Is there a way I can modify the java code once I install the project through NPM? because I could try to add support to it.

rusel1989, 2016-11-25:
The best way to do this, is to clone this repository, fiddle with code and then create a pull request, so the others can use it as well.

ErreMalote, 2016-11-25:
After doing some investigation I figured that the device I am using requires BluetoothGatt to connect to read from UART notifications. In your project you are using sockets. I believe that BluetoothGatt can connect to any device. Anyway, thanks for your responses I will probably would have to start a new project or look for something that is already started on that.

rusel1989, 2017-01-12:
@nehasalian @wzh19960613 It seems that its related to api level 23 permissions, do you both use api level 23 ? as @ErreMalote  does ...

wzh19960613, 2017-01-14:
@rusel1989 I have no question here now because I found the devices which I need connect were BLE devices... So I have used another solution.

berkeguncan, 2018-04-04:
discoverUnpaired = () => {
    if (this.state.discover) {
      console.log('Hey')
      return false
    } else {
      console.log('Begin')
      this.setState({ discover: true })
      BluetoothSerial.discoverUnpairedDevices()
      .then((unpairedDevices) => {
        this.setState({ unpairedDevices, discover: false })
        console.log(this.state.unpairedDevices)
      })
      .catch((err) => alert('Error'))
    }
  }

i'm trying log the unpaired devices but it still returns empty array.I'm gonna go crazy.My deadline is so near and i must solve this problem please HEEEELP :'(

gabrielhpugliese, 2018-08-08:
Hi guys, I came from the future to ask how did you fix this issue.

lukadumancic, 2019-03-08:
Coming from far future,
Just add this lines to AndroidManifest 
```
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```
After that you'll have to go to app settings and allow that permission manually, or you can follow [ReactNative official docs](https://facebook.github.io/react-native/docs/permissionsandroid)
I hope this helps

rishaChoubey, 2019-10-21:
Even after adding 
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
these two lines in manifest file I'm unable to get the list of unpaired devices.

Can someone help?!
    /**
   * [android]
   * Discover unpaired devices, works only in android
   */
  discoverUnpaired () {
    if (this.state.discovering) {
      return false
    } else {
      this.setState({ discovering: true })
      BluetoothSerial.discoverUnpairedDevices()
      .then((unpairedDevices) => {
        alert(unpairedDevices)
        this.setState({ unpairedDevices, discovering: false })
      })
      .catch((err) => Toast.showShortBottom(err.message))
    }
  }
 Am i making any mistake?

ChiragDugar, 2020-04-06:
Hey!

The possible problem is that you have not allowed permissions for using GPS. Using Bluetooth also requires the GPS permission. 

Go to Apps in your phone and check if location permissions are given or not. 
If not, then write the corresponding code in the componentDidMount() function and allow permission as soon the user opens the app. 

Hope this works :)

ProgPhil, 2022-04-09:
Going into app setting and manually giving it all permissions, especially Locations, solved the issue for me.