rusel1989/react-native-bluetooth-serial #11: Connect to Arduino with bluetooth serial port module

techniq, 2016-11-15:
I'm looking for a solution to connect to an Arduino running Firmata using a bluetooth serial port module, such as the [JY-MCU](https://www.amazon.com/Phantom-YoYo-Arduino-Bluetooth-Wireless/dp/B009DZQ4MG/ref=sr_1_8?ie=UTF8&qid=1479187266&sr=8-8&keywords=bluetooth+serial).  It appears this project would work for this, but I'm having trouble understanding how to make that work.  

Using [Johnny-five](http://johnny-five.io/api/board/), you can pass a `node-serialport` instance, but i'm not sure if this offers a drop-in replacement.  I've seen other projects like https://github.com/jacobrosenthal/react-native-ble and https://github.com/elin-moco/ble-serialport that also look promising, but they are only for BLE and not full-fledged Bluetooth, which I think I need for the hardware module.

I have a PoC connecting to the the device from my Mac, but am struggling to find how to accomplish the same from React-Native (since `/dev/*` isn't a concept there).

rusel1989, 2016-11-30:
This issue seems to be related to BLE on android which is not supported .. so if u manage to solve it just send PR .. For now I am closing this issue

closed 2016-11-30 by rusel1989

techniq, 2016-11-30:
@rusel1989 This was actually asking about Bluetooth Classic on Android (the JY-MCU is a Bluetooth Classic serial port).  On my mac after I pair it creates a serial device under /dev that I use the port and was trying to accomplishing something similar using React Native.