rusel1989/react-native-bluetooth-serial #13: TypeError: Cannot set property 'on' of undefined

timscott, 2016-11-29:
I'm getting the same error as issue https://github.com/rusel1989/react-native-bluetooth-serial/issues/6.

```
TypeError: Cannot set property 'on' of undefined
    at http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=true&minify=false:138348:19
    at loadModuleImplementation (http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=true&minify=false:127:1)
    at guardedLoadModule (http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=true&minify=false:70:8)
    at _require (http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=true&minify=false:54:1)
    at http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=true&minify=false:136436:33
    at loadModuleImplementation (http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=true&minify=false:127:1)
    at guardedLoadModule (http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=true&minify=false:70:8)
    at _require (http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=true&minify=false:54:1)
    at http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=true&minify=false:94598:28
    at loadModuleImplementation (http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=true&minify=false:127:1)
```

Initially, I was missing necessary code in ```MainApplication.java```. After I added it, I continue to get the same error. 

I'm on ```react-native@0.30.0-rc.0```. Could that be the problem? Ideas?

rusel1989, 2016-11-30:
For me it seems like some issue with cache ... also please specify platform where you are experiencing this issue

timscott, 2016-11-30:
@rusel1989 I'm running on an Android device. I tried ```react-native start --reset-cache``` but still get the error.

timscott, 2016-12-03:
I have upgraded to ```react-native@0.39.0```. Same error.

timscott, 2016-12-06:
Got it working. Not sure what I did.

closed 2016-12-06 by timscott

rusel1989, 2016-12-06:
Okay cool ... I am really not sure how this can be happening