rusel1989/react-native-bluetooth-serial #16: Can read data from bluetooth?

zhanguangao, 2016-12-07:


dalmago, 2017-02-20:
You can read using readFromDevice:

`BluetoothSerial.readFromDevice().then((data) => {console.log(data)});`

joshuapinter, 2017-05-23:
Do you have to run this with a `setTimeout` to get updated data from the device?

prashant45, 2017-08-08:
How can I read continuously from the bluetooth device?

joshuapinter, 2017-08-08:
@prashant45 I'd much rather get an event listener setup but I ended up using a `setInterval()` as a quick and dirty way to keep reading the latest data.

prashant45, 2017-08-08:
@joshuapinter Thanks that worked for now 👍

prashant45, 2017-08-16:
@joshuapinter did you manage to set up the event listener?

joshuapinter, 2017-08-16:
@prashant45 No. Still using `setInterval()`.

prashant45, 2017-08-16:
perhaps this could be of help [react-native-event-listeners](https://github.com/meinto/react-native-event-listeners).

joshuapinter, 2017-08-16:
@prashant45 It'll be relatively easy to do but it's just not necessary for what we're doing. The setInterval works fine for now. This isn't my repository but you should be able to get a working example going quite easily. I believe the library even supports it but it's not very well documented.

prashant45, 2017-08-16:
Yes, the library is not well documented. For me, ``setInterval()`` is not working in a long run. I also need to do a lot of stuff after  ``BluetoothSerial.readFromDevice().then((data) => {...});`` and app stops responding probably because it does not complete doing the task in assigned interval of 2 seconds.

andrescheca, 2017-10-24:
Hi, I agree, the library is not very well documented but I was reading the Java code and found that `BluetoothSerial.withDelimiter` exists. I used it with the carriage return character `\r` and `BluetoothSerial.on('read')` now works!: 
```
BluetoothSerial.withDelimiter('\r').then(() => {
      Promise.all([
        BluetoothSerial.isEnabled(),
        BluetoothSerial.list(),
      ]).then(values => {
        const [isEnabled, devices] = values;
        ...
      });
      BluetoothSerial.on('read', data => {
        console.log(`DATA FROM BLUETOOTH: ${data.data}`);
     });
});
```
That way you don't have to use `setInterval()` for reading data.

wenchao1020, 2019-07-11:
I can not read hex data in normal mode, but in debugger mode I can do it. change the function in `RCTBluetoothSerialModule.java` will fix it

if use `withDelimiter ` change the `onData` function like that

`void onData (String data) {
        mBuffer.append(data);
        String completeData = readUntil(this.delimiter);
        if (completeData != null && completeData.length() > 0) {
            WritableMap params = Arguments.createMap();

            WritableArray writableArray = Arguments.createArray();
            byte[] bytes = completeData.getBytes();
      
            for(int i=0;i<bytes.length;i++){
                writableArray.pushInt((int)bytes[i]);
            }

            params.putArray("arr", writableArray);
            params.putString("data", completeData);
            sendEvent(DEVICE_READ, params);
        }
    }`
and in you `js` file,  use ` Buffer.from(data.arr)` get string data . 

if `readFromDevice` change `readFromDevice` like that 
`public void readFromDevice(Promise promise) {
        if (D) Log.d(TAG, "Read");
        int length = mBuffer.length();
        String data = mBuffer.substring(0, length);
        mBuffer.delete(0, length);
        
        WritableArray writableArray = Arguments.createArray();
        byte[] bytes = data.getBytes();
        for(int i=0;i<bytes.length;i++){
            writableArray.pushInt((int)bytes[i]);
        }

        promise.resolve(writableArray);
    }`
it will get the Buffer array, not the string data. need to use ` Buffer.from(data)`

Rehan-Rehman-Punjwani, 2020-02-23:
> Hi, I agree, the library is not very well documented but I was reading the Java code and found that `BluetoothSerial.withDelimiter` exists. I used it with the carriage return character `\r` and [...]

> Hi, I agree, the library is not very well documented but I was reading the Java code and found that `BluetoothSerial.withDelimiter` exists. I used it with the carriage return character `\r` and [...]

@andrescheca I am following the same as you have used but event listener is not firing is not working in my case. Kindly help me ..

andrescheca, 2020-02-24:
Hey @Rehan-Rehman-Punjwani, it's been a long time since I used this library but I can try to help you. Are you able to connect to the device within your React Native app? Have you tried using an app like Bluetooth Terminal in Android to read and send data to the device?

Rehan-Rehman-Punjwani, 2020-02-24:
@andrescheca Thank you for your reply,  yes I have..  it is working fine with reading and writing data to the android terminal!

andrescheca, 2020-02-24:
What about connecting to the bluetooth device from your React Native app?

Rehan-Rehman-Punjwani, 2020-02-24:
@andrescheca Yes, it is working.

andrescheca, 2020-02-24:
Have you tried testing it in debugger mode as @wenchao1020 suggested and check if you are receiving data in the Java's onData method?

Rehan-Rehman-Punjwani, 2020-02-27:
> > Hi, I agree, the library is not very well documented but I was reading the Java code and found that `BluetoothSerial.withDelimiter` exists. I used it with the carriage return character `\r` and `BluetoothSerial.on('read')` now works!:
> > ```
> > BluetoothSerial.withDelimiter('\r').then(() => {
> >       Promise.all([
> >         BluetoothSerial.isEnabled(),
> >         BluetoothSerial.list(),
> >       ]).then(values => {
> >         const [isEnabled, devices] = values;
> >         ...
> >       });
> >       BluetoothSerial.on('read', data => {
> >         console.log(`DATA FROM BLUETOOTH: ${data.data}`);
> >      });
> > });
> > ```
> > 
> > 
> > That way you don't have to use `setInterval()` for reading data.
> 
> > Hi, I agree, the library is not very well documented but I was reading the Java code and found that `BluetoothSerial.withDelimiter` exists. I used it with the carriage return character `\r` and `BluetoothSerial.on('read')` now works!:
> > ```
> > BluetoothSerial.withDelimiter('\r').then(() => {
> >       Promise.all([
> >         BluetoothSerial.isEnabled(),
> >         BluetoothSerial.list(),
> >       ]).then(values => {
> >         const [isEnabled, devices] = values;
> >         ...
> >       });
> >       BluetoothSerial.on('read', data => {
> >         console.log(`DATA FROM BLUETOOTH: ${data.data}`);
> >      });
> > });
> > ```
> > 
> > 
> > That way you don't have to use `setInterval()` for reading data.
> 
> @andrescheca I am doing the same as you have done but still, the event listener is not firing in my case. Kindly help.

@andrescheca can you please tell me that how do i know the data i am sending have any delimiter in it or not. Because i am just sending a string like "Hello" from the Linux server to the app..?

Rehan-Rehman-Punjwani, 2020-02-27:
> Have you tried testing it in debugger mode as @wenchao1020 suggested and check if you are receiving data in the Java's onData method?

@andrescheca thank you for the help. I have just solved the issue. I added '\r' at the end of data that i was sending and it worked.

andrescheca, 2020-02-27:
@Rehan-Rehman-Punjwani That's really great!

falcon977, 2021-06-14:
Hello there, I have the same problem, I am creating  a react native chat application that communicates with a HC05 bluetooth module connected to an STM32, and my main problem is to receive instant messages from the module. I already read all your interactions in this forums and tried using the event listener provided by the library but it doesn't trigger instantly the messages, and when it triggers, it duplicates the messages multiple times ? Any solution please ?

taj567, 2022-01-04:
Hello Guys, if you have any idea how to retrieve the data from a Bluetooth paired device(ex: as we connected to a Philips CPAP machine) in react-native

taj567, 2022-01-05:
> BluetoothSerial
am able connect device but not able to retrieve the data