rusel1989/react-native-bluetooth-serial #17: Unable to Connect to Paired Device

timscott, 2016-12-07:
Not sure if this is a bug or I am doing something wrong. Basically I am unable to connect with an already paired device.

I have code like this:

```javascript
return BluetoothSerial.isConnected().then(isConnected => {
  if (isConnected) {
    return true;
  }
  return BluetoothSerial.connect(lmu.address).then(response => {
    return response;
  });
});
```

The first time this code runs ```connect``` is called, and the (Android) phone prompts me to accept the pairing. I accept. Connect succeeds. I am able to ```write``` and ```read```. All is well.

Upon subsequent passed through this code, ```isConnected``` is true. Again, all is well.

Then I close and restart the app. After that when this code runs, I get an error on ```connect```: "Error: Unable to connect to device"

Then I on the phone go to Settings > Bluetooth and Unpair from the device. Now when this code runs ```connect``` succeeds again.

I tried skipping ```connect``` if the device is already paired and just proceed to ```write```. I did not expect that to work, and it does not.

rusel1989, 2016-12-09:
@timscott Hey, that is very weird, I didn't have this issue before, but maybe the connection isn't terminated properly, can you try calling disconnect in componentWillUnmount or somewhere similar ?

timscott, 2016-12-28:
@rusel1989 I have narrowed this down a little. It only happens upon pressing ```Reload``` on the dev menu. It does not happen when stopping or killing the app on the device. Unfortunately, reloading apparently does not trigger ```componentWillUnmount``` or any other event handler I can find. [I've posed this question on StackOverflow.](http://stackoverflow.com/questions/41365123/reload-leaves-corrupt-state)

I wonder if there is any way to handle it in this component? Perhaps the fact that is does not happen when the app closes, but does happen when JS is  reloaded offers some clue?

I suppose it's development-only problem, however, it's a quite painful one.

rusel1989, 2017-01-12:
Hey this seems to me same as @waltermacambira describes in https://github.com/rusel1989/react-native-bluetooth-serial/issues/15 so i will close this, the fix is almost ready :-)

closed 2017-01-12 by rusel1989

ramacha7, 2021-02-26:
@rusel1989 I am facing similar problems. I discover the paired devices in the useEffect() and store the paired devices. Then I have a button that when I press should connect to the device that I want to connect to. Sometimes when I re-run the app, it is suddenly not able to connect to the device. It happens at random times. I was wondering if you could help with this.

zainulabideen-dev, 2022-07-24:
Hello guys,
I don't come up with a solution but I want to share one thing here. So when I try to connect my mobile device with my LapTop Bluetooth It connected successfully with the same function. But I'm not able to connect to another mobile device.