rusel1989/react-native-bluetooth-serial #21: Cant run example

diogoviannaaraujo, 2017-01-30:
My example does not run, after run-ios i get a crash:

> An error was encountered processing the command (domain=NSPOSIXErrorDomain, code=2):
Failed to install the requested application
An application bundle was not found at the provided path.
Provide a valid path to the desired application bundle.
Print: Entry, ":CFBundleIdentifier", Does Not Exist

TylerWilcock, 2017-02-02:
I'm encountering the same issue.

m7775223, 2017-04-13:
You can try ‘react-native upgrade’.I use this command to solve the problem

rusel1989, 2017-04-13:
I have been developing mostly for android for last couple of months, so iOS version might be buggy. 
Unfortunately I am pretty busy so PRs are welcomed 😉

LingboTang, 2017-12-12:
@aduroidpc I used that command to solve the problem, however, the example posted here is still outdated, it does not work with latest version of react-native 0.51, we can build it, but we can't successful run it on devices. 

@rusel1989 , there are couple of things you probably need to update for your example. 

1. You need to change the route of your application in `index.js`, you are importing `App` from `./src/App.js` which should be `BluetoothSerialExample`.

2. NativeModule cannot resolve `BluetoothSerial`, I haven't figure out what the real cause is, so please look at it. 

I would like to use bluetooth serial once its up-to-date again. I've used the phoneGap version before for Android, it works well. I would use this for ios as well.