rusel1989/react-native-bluetooth-serial #28: How to connect two BT devices, looks like no!

liyatang, 2017-03-07:


prashant45, 2017-09-18:
Even I am having problems connecting 2 bluetooth devices?

Any alternatives for this issue ?

liyatang, 2017-09-18:
react-native-bluetooth-serial is singleton，you can fork it then modify。   

eg:https://github.com/gmfe/react-native-gm-bluetooth

alanyong91, 2018-03-25:
Actually can. You just need write more code.

arevaldez, 2019-09-23:
Any idea how to implement this? I'm trying to implement this but it keeps disconnecting from the first device after I try connecting to the second device. @liyatang @alanyong91

alanyong91, 2019-09-23:
The way I did is connect the first device and disconnect, then connect the other one. I used this package to connect bluetooth printer. So might be different approach with you.

alex-valdez, 2019-09-23:
@alanyong91 were you able to connect to two devices at once? Or just one device? I am tying to connect the phone to two devices at once. Thanks for your help!

alanyong91, 2019-09-24:
I can connect two devices at once but only one will use if call both devices. I create a class (eg: BluetoothConnect) to call `bluetoothSerial` and use `new BluetoothConnect()` for each device.