rusel1989/react-native-bluetooth-serial #32: Pairing rejected by UART

bmoe872, 2017-04-01:
First just wanted to say thank you so much to the contributors working on this. 

# Specs
I am using the BluetoothSerialExample you have provided.
I am using this bluetooth module: https://www.adafruit.com/product/1697
with the "EchoDemo" found here: https://learn.adafruit.com/getting-started-with-the-nrf8001-bluefruit-le-breakout/software-uart-service

I am running the example app on a Sony Xperia Z3v running android 5.1.1. 

# My issue
I have the app running on the phone, and am able to discover the bluetooth connected to my Arduino. However everytime I try to connect I am getting a message on the app `Pairing rejected by UART` on the serial monitor through the arduino IDE, I am seeing this:
[image]

It connects for a short bit and then goes back to advertising. I am unsure of what could be causing this, and I tried to look through your code to understand what could be causing this, but to no avail. 

Am I missing something on the arduino sketch end or is there something more that I should be doing inside the example sketch? 

Thanks for any help, and again love the library. I hope to be able to help in anyway I can once I get the example working.

rusel1989, 2017-04-13:
Hey @bmoe872 Bluetooth Low Energy is not supported on android, and probably won't be for a while 😕

bmoe872, 2017-04-13:
Hey @rusel1989 thanks so much for getting back to me! Again love what you're doing. BLE isn't supported on android? I must be confused then how the Adafruit app works? I've gotten the BT module Im using to work with Adafruit's app:
https://play.google.com/store/apps/details?id=com.adafruit.bluefruit.le.connect&hl=en

Am I missing something with that?

rusel1989, 2017-04-13:
@bmoe872 Sorry, my previous reply is poorly formulated, What I meant is that this library doesn't support Bluetooth LE on Android 😒

bmoe872, 2017-04-13:
@rusel1989 ahhh I see. Well then. Thank you for letting me know. I'll have to switch out my BT module! 🔨

closed 2017-05-31 by bmoe872