rusel1989/react-native-bluetooth-serial #33: A doubt about this tool

costabatista, 2017-04-04:
Does This tool serve to send archives for impression in a thermal printer?

rusel1989, 2017-04-13:
@costabatista Sorry I dont know what "Send archives for impression" means ? 

Anyway I am using it to print receipts on thermal printers without any difficulties both ios and android.

costabatista, 2017-04-13:
Sorry about my bad english, I wanna mean "Send stream of bytes to thermal printers". So, later some number of tentatives i did a bluetooth connection using your implementation. Thanks! =)

rusel1989, 2017-04-13:
:+1: I am glad to hear that, I am closing this issue for now, feel free to reopen if you need some further info ...

closed 2017-04-13 by rusel1989

pewh, 2017-06-15:
@rusel1989 would you mind if you inform me (or give a link) about what data need to send to print receipts via thermal printers? My use case is send invoice (also contain image & qr code). Thanks.