rusel1989/react-native-bluetooth-serial #34: [ios]undefined is not an object(evaluating'BluetoothSerial.on....')

m7775223, 2017-04-13:
I run this example in IOS, but it appears the following error, I can't solve it, do you know why?

undefined is not an object(evaluating'BluetoothSerial.on=function(eventName,handler){
deviceEventEmitter.addListener(eventName,handler);
}')

AlexKotov, 2017-06-29:
Same problem.

react-native-cli: 2.0.1
react-native: 0.44.0
node: 8.1.2
xcode: 8.3.3

Senocico, 2017-08-02:
the same issue...

nicoceledon, 2017-10-04:
I have the same problem. Any clue?

sadidebbal-rhizom-team, 2018-03-06:
Hello, I have the same problem on android with:
`"react-native": "0.50.3",
"react-native-bluetooth-serial": "^1.0.0-rc1",
`
Does anyone got a solution for this ?

Gomechatronics, 2019-05-09:
I have the same issue. Anyone who managed to solve it?

AmmarIsmaeel, 2019-07-16:
`react-native link` will solve the problem!

nicoceledon, 2019-07-24:
> I have the same issue. Anyone who managed to solve it?

On latest version of rn i haven't seen that error again. And i don't remember how did i solved previously.

canthony224, 2020-01-08:
I know this is a two year longshot, but does anyone have (and remember) a fix to this?

iamandiradu, 2020-01-13:
Trying to find the same thing, man. @canthony224

lchenfox, 2020-08-09:
The same issue. Is this lib still maintained?

smkjason, 2021-02-16:
Anything...?