rusel1989/react-native-bluetooth-serial #37: Listener for receiving data

happierco, 2017-04-24:
I need to read from a barcode scanner without using an input field. Is there a listener I can use?

rusel1989, 2017-04-25:
Hey @happierco , You can use `BluetoothSerial.on('data', (data) => ...do something with data)`

ibenben, 2017-05-05:
`BluetoothSerial.on('data', (data) => {
        console.log(data);
 });`

Can not receive data.

ibenben, 2017-05-05:
Hao can I read the byte from device.

babyinthesky, 2017-06-19:
I get the listener for receiving data working by the following way. Note that you have to replace the delimiter '\r\n' with the one upon your own case.

BluetoothSerial.withDelimiter('\r\n').then((res)=>{
console.log("delimiter setup",res);
BluetoothSerial.on('read',(data)=>{
console.log('read',data);
})
})

goldfishinsky, 2019-03-04:
> I get the listener for receiving data working by the following way. Note that you have to replace the delimiter '\r\n' with the one upon your own case. [...]

where should I put these codes?

hungtm84, 2019-09-10:
> BluetoothSerial.withDelimiter('\r\n').then((res)=>{
> console.log("delimiter setup",res);
> BluetoothSerial.on('read',(data)=>{
> console.log('read',data);
> })

I have same question, did @goldfishinsky solve this problem ?

yalcinozer, 2020-10-08:
> I get the listener for receiving data working by the following way. Note that you have to replace the delimiter '\r\n' with the one upon your own case. [...]

I know this issue is a little old but,  Should we place on("data") listener in "withDelimiter" method

All I want actually, listening constantly for data from HC05 module to app. Only thing I can do is checking module manually with readFromDevice method.

ramacha7, 2021-02-25:
@yalcinozer Have been able to figure out how to do it. I actually have the same application goal. I need to constantly listen for data from a HC05 module.

yalcinozer, 2021-02-25:
@ramacha7 using delimiters is a must be. You also should call functions repeatedly to read data. If you have any option use "BLE" devices. At some point I just gave up checking if data received or not.