rusel1989/react-native-bluetooth-serial #40: RFCOMM Support

axtscz, 2017-05-09:
Will this library connect an Android phone and an RFCOMM server?

mvellon, 2017-09-05:
Is there anything magical about that UUID (in the Pi python app)?  I have a very similar example but I can't get the library to list() my device.  My own service uses the SERIAL_PORT class and profile (it's based on the same example).

storerjeremy, 2017-09-06:
@mvellon I don't think there is anything magical about the UUID you should be able to use any. But to get my pi python app to work, be discoverable and use SERIAL_PORT I had to do a few things.

First I had to run the bluetooth daemon in compatibility mode by adding a "-C" after the bluetoothd line in /etc/systemd/system/dbus-org.bluez.service and then enable serial profile by running "sudo sdptool add SP"

Then to make the device visible run "sudo hciconfig hci0 piscan"

I got this info from [here](https://github.com/EnableTech/raspberry-bluetooth-demo/blob/master/README.md) and [here](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=133263)

Hope that helps.

mvellon, 2017-09-06:
Yep, I've done all that. I can connect to it from another Pi, but not iOS. I will continue to experiment. 

Thanks for the quick response tho!

storerjeremy, 2017-09-06:
No worries. I've only used it with Android so far. I feel like SPP may not be supported on IOS

storerjeremy, 2017-09-11:
@mvellon Did you end up getting it to work with IOS?

mvellon, 2017-09-11:
No.  I think that iOS has limited support for classic Bluetooth. If you’re not a “blessed” iOS perisperhal, you generally need to go the BLE route with GATT advertisements/services/characteristics.

mvellon, 2017-09-11:
Oh, and iOS doesn’t seem to support RFCOMM in its BLE stack.

pstanton, 2017-11-19:
what do i have to do to connect RFCOMM correctly? my remote RFCOMM service works fine with other bluetooth terminal apps ([such as](https://play.google.com/store/apps/details?id=com.dev.aproschenko.bluetoothterminal&hl=en)) however, after adding a TextInput to the example app, and then calling `BluetoothSerial.write(text)`, nothing gets through to the other end... help appreciated.

storerjeremy, 2017-11-19:
@pstanton `BluetoothSerial.write()` should definitely work. Is your RFCOMM server expecting a delimiter such a \n? Give [this RFCOMM server](https://github.com/kuri65536/python-for-android/blob/master/python-modules/pybluez/examples/simple/rfcomm-server.py) a go.

pstanton, 2017-11-20:
i'll try appending "\r\n" .. like i said the server seems to work with other android terminals so was expecting it to be easy. there's no character encoding necessary?

also, have you had any luck with iOS? using BLE perhaps? we're trying to BLE/uart/serial if RFCOMM doesn't work...

pstanton, 2017-11-20:
ah, the line break! that works .. for the TX anyway. I cant seem to read data, i have set up listeners:

		```
BluetoothSerial.on("connectionLost", () => this.onConnectionLost);
		BluetoothSerial.on("data", data => {
			console.log("data", data);
		});

		BluetoothSerial.withDelimiter("\n").then(res => {
			console.log("delimiter setup", res);
			BluetoothSerial.on("read", data => {
				console.log("data", data);
			});
		});
```

but nothing is logged

storerjeremy, 2017-11-20:
I am using essentially the python RFCOMM server I posted above and all I am doing in my RN app after connecting is setting up a listener with a newline delimiter, very similar to you and this all works fine for me.

`BluetoothSerial.on('read', (data) => { console.log(data); });`
`BluetoothSerial.withDelimiter('\n').then(() => { console.log('Bluetooth subscribed with delimiter'); });`

I use text written to my bluetooth server to stop/start my sending of data. ie.

`BluetoothSerial.write('start');`

Then as the RN app is listening for a newline delimiter from my RFCOMM server I am sending my data with a \n appended to it ie.

`client_sock.send("some data to send to react native from rfcomm server\n")`

Regarding using IOS and RFCOMM I haven't tried doing it myself yet but I did find [this issue](https://github.com/rusel1989/react-native-bluetooth-serial/issues/3#issuecomment-238252084) talking about adding in your specific service UUID in the IOS code. I will be testing this out soon once an iphone arrives in the mail.

sadidebbal-rhizom-team, 2018-04-20:
Hello, I'm trying to display timeseries from my raspberry to my react-native app. The connexion is done between the two devices, but i don't know how to display it.

Satyam-code143, 2021-03-19:
@storerjeremy Hey, I am trying to communicate with the raspberry pi with my RN application over a serial connection but i am not getting the pairing and connecting part of the RN code. I have a similar code of the for the RFCOMM server but still I am not able to find the solution. Can u please help me out with the RN side code.

Thank you!!

Satyam-code143, 2021-03-19:
@pstanton Hey, did the `BluetoothSerial.write()` work?? Please help me with that!!