rusel1989/react-native-bluetooth-serial #43: Android 7.0 not supported

rexlow, 2017-05-20:
Tested on both of my Galaxy S3 (v5.1) and Galaxy Note 5 (v7.0).

The app in my Note 5 won't be able to discover devices.

joshuapinter, 2017-11-21:
Same here. I know there were some big changes to permission in Android 6 that we need to handle in order for this library to discover devices, but I'm not sure what's changed with Android 7 that requires some changes.

busy-banana, 2017-12-14:
@rexlow @joshuapinter 
You can try to add the following code to AndroidManifest.xml in your project and open the positioning permissions. It worked on my XIAOMI 5X with Android 7.1.2 

uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>