rusel1989/react-native-bluetooth-serial #45: Example not working

jchirschy, 2017-05-29:
Following the install procedures, I've got the following error at runtime : 
```javascript
(node:14363) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 2): TypeError: Cannot read property 'message' of undefined
```
Any idea what's going wrong ?

bmoe872, 2017-05-29:
Have you ran 'react-native link' after installing the library?

jchirschy, 2017-05-29:
yes.

I managed to launch the example.

First of all, it was actually a **resolving dependencies issues** : 
`"com.atlassian.mobile.video" is not avaible on maven right now.`
So I had to update react and reactnative just as explain here : https://stackoverflow.com/a/44216865
by editing `package.json`
```
"react": "16.0.0-alpha.3",
"react-native": "0.43.1",
```
and then running `npm install --force`

And then, I've got another issue : 
`cannot override getUseDeveloperSupport() as protected`
which was resolved by following this https://github.com/airbnb/lottie-react-native/issues/38
which consisted to edit `BluetoothSerialExample/android/app/src/main/java/com/bluetoothserialexample/MainApplication.java` and set `getUseDeveloperSupport`to public.