rusel1989/react-native-bluetooth-serial #55: react-native-bluetooth-serial:compileReleaseJavaWithJavac FAILED

InevitableAnup, 2017-08-16:
Hi I am trying to integrate react-native-bluetooth-serial into my react-native app.I followed the documentation and completed these 3 steps:-
1)Install package via npm: npm i -S react-native-bluetooth-serial
2)Link libraries with: rnpm link or react-native link for React Native >= 0.27
3)For android you also need to put following code to AndroidManifest.xml
<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />

After this I ran react-native run-android and the app started failing with the following error:-
1 error
:react-native-bluetooth-serial:compileReleaseJavaWithJavac FAILED

Please revert as soon as possible.
Thank you.

sn1k, 2017-08-23:
Same problem here.

sn1k, 2017-08-23:
Since RN >=0.47 you should delete the @override into  package
`   @Override
    public List<Class<? extends JavaScriptModule>> createJSModules() {
        return Collections.emptyList();
    }` 
Change for 
`  public List<Class<? extends JavaScriptModule>> createJSModules() {
      return Collections.emptyList();
  }`

tofoli, 2017-10-17:
Can someone make the same change done [here](https://github.com/react-native-community/react-native-blur/pull/226/files/a93b304d9aea69d8922b3be9c6c3ad560616d6c2?diff=unified)

pstanton, 2017-11-19:
not sure what has to be done to get this into the npm system ? no changes to this project in 9+months...