rusel1989/react-native-bluetooth-serial #57: Device list always returns empty

MagganKoolman, 2017-10-09:
I must have missed an essential part for this module. Whenever discoverUnpairDevices is called it returns an empty list. No error messages or warnings are thrown. I used the following code.

```
import BluetoothSerial from 'react-native-bluetooth-serial'

BluetoothSerial.enable()
.then((res) => console.log(res))
.catch((err) => console.log(err.message))

BluetoothSerial.discoverUnpairedDevices()
.then((unpairedDevices) => {
  console.log(unpairedDevices)
})
.catch((err) => console.log(err.message))
```

closed 2017-10-10 by MagganKoolman

natrey, 2017-10-25:
I was trying to run the example app on android 6.0 and did also receive an empty list. In my case, i had to add additional config lines to AndroidManifest.xml
```
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```
as mentioned [here](https://stackoverflow.com/a/36164455/5899355)

joshuapinter, 2017-11-22:
More than that, you need to ask for permissions in the Activity as well. The SO answer you referenced shows how to do that.