rusel1989/react-native-bluetooth-serial #58: How to send write image

Raveerote, 2017-10-13:
I'm trying to write image but they all didn't work. Here's the code

1.
 var byteArray = Base64.decodeString(base64String); 
                    for (var i = 0 ; i < byteArray.length; i++){
                        this.write(new Buffer(byteArray[i].toString(2)))
                    }

2. 
this.write(byteArray.buffer)

3. 
 var byteArray = Base64.decodeArrayBuffer(base64String); 
                    console.log(byteArray)
                    this.write(byteArray)

NativoLink, 2017-12-22:
Go to http://www.zjiang.com/en/init.php/service/driver and Download  -> **SDK for Android OS**
 in this you will find  -> **btsdk.jar** copy this and create the libs folder in  your /node_modules/react-native-bluetooth-serial/android/libs and **paste**, add this line **compile files('libs/btsdk.jar')**  in  /node_modules/react-native-bluetooth-serial/android/builde.gradle.
**Add this lines in RCTBluetoothSerialModule** 

 private Bitmap base64ToBitmap(String b64) {
        byte[] imageAsBytes = Base64.decode(b64.getBytes(), Base64.DEFAULT);
        return BitmapFactory.decodeByteArray(imageAsBytes, 0, imageAsBytes.length);
    }

    private String bitmapToBase64(Bitmap bitmap) {
        ByteArrayOutputStream byteArrayOutputStream = new ByteArrayOutputStream();
        bitmap.compress(Bitmap.CompressFormat.PNG, 100, byteArrayOutputStream);
        byte[] byteArray = byteArrayOutputStream .toByteArray();
        return Base64.encodeToString(byteArray, Base64.DEFAULT);
    }

    @ReactMethod
    /**
     * Write to device over serial port
     */
    public void writeBytes(String imgB64, Callback callback) {
        String encodeIMG = imgB64;
        String base64Image = encodeIMG.split(",")[1];
        Bitmap bmp = base64ToBitmap(base64Image);
        String encodedImage = bitmapToBase64(bmp);
        //---------------------------------------------------
        byte[] data = POS_PrintBMP(bmp, 384, 0);  // --- PRINT IMG FROM BITMAP
        //---------------------------------------------------
        mBluetoothService.write(data);
        callback.invoke(true);
    }

public static byte[] POS_PrintBMP(Bitmap mBitmap, int nWidth, int nMode) {
        int width = (nWidth + 7) / 8 * 8;
        int height = mBitmap.getHeight() * width / mBitmap.getWidth();
        height = (height + 7) / 8 * 8;
        Bitmap rszBitmap = mBitmap;
        if(mBitmap.getWidth() != width) {
            rszBitmap = Other.resizeImage(mBitmap, width, height);
        }

        Bitmap grayBitmap = Other.toGrayscale(rszBitmap);
        byte[] dithered = Other.thresholdToBWPic(grayBitmap);
        byte[] data = Other.eachLinePixToCmd(dithered, width, nMode);
        return data;
    }

RaiiZen, 2017-12-22:
@NativoLink can you more details how add this ?

**private Bitmap base64ToBitmap(String b64) {
byte[] imageAsBytes = Base64.decode(b64.getBytes(), Base64.DEFAULT);
return BitmapFactory.decodeByteArray(imageAsBytes, 0, imageAsBytes.length);
}**

or can you give me full code RCTBluetoothSerialModule , please i need this.

getabatu, 2018-01-13:
how do you send a text into thermal printer using write method

Kaitsuvaan, 2018-11-13:
@NativoLink thx for code 👍 

Must add:
@qcxiaoshuai 
import android.graphics.Bitmap;
import android.graphics.BitmapFactory;

import java.io.ByteArrayOutputStream;
import zj.com.customize.sdk.Other;

with **RCTBluetoothSerialModule.java** too and its work fine :)

renrukun861113, 2019-01-13:
This is just for Android's solution. How should IOS be solved? It's urgent. Thank you!

MuhammadSNajib, 2019-01-25:
> This is just for Android's solution. How should IOS be solved? It's urgent. Thank you!

+1

dedenhendrap, 2019-05-31:
how to use this function ? please

zainmustafa, 2019-08-19:
@NativoLink thanks for this, able to print images in some thermal printers but some aren't printing even with this, any idea? And how can i solve this?

FazilMuhammed, 2020-10-13:
@renrukun861113 @msnajib hello guys did you find the solution how to print Image in iOS ?