rusel1989/react-native-bluetooth-serial #64: Write Method on this component

getabatu, 2018-01-11:
hi i wanted to know how to run a write method on this component? how to send blue tooth data to thermal printer?

thank you in advance

mthrsj, 2018-12-18:
To write data to any component you can use the method `write()`, like:

```javascript
// ...
import BluetoothSerial from "react-native-bluetooth-serial";

const MySerial = new BluetoothSerial();

// Then, you need to connect to the desired device, through the id.

MySerial.connect("XX:XX:XX:XX") // MAC address

MySerial.write("My message")

```

Don't forget to connect to the device before trying to send any message. Hope it helps.

chamaloriz, 2019-09-29:
let encoder = new EscPosEncoder();

    let result = encoder
      .initialize()
      .text("HelloWorld")
      .newline()
      .qrcode('HelloWorld')
      .encode();

    const { Buffer } = require('buffer');

    BluetoothSerial.write(Buffer.from(result.buffer))