rusel1989/react-native-bluetooth-serial #67: This plugin does not work!!!!

ithieund, 2018-02-02:
@rusel1989 ,
Is this project continue maintaining?
I tried to install and run but it throw an error:

`/Users/hieunguyen/Mobile/ArduinoBT/node_modules/react-native-bluetooth-serial/android/src/main/ja
va/com/rusel/RCTBluetoothSerial/RCTBluetoothSerialPackage.java:23: error: method does not overrid
e or implement a method from a supertype
    @Override
    ^
1 error
:react-native-bluetooth-serial:compileReleaseJavaWithJavac FAILED`

I also tried dowloading the example project and run but no luck too.
It throw error about 'sane' while running react-native run-android

Can you please fix these issue?
Thanks.

darkrift, 2018-02-04:
The method that is being overridden has been deprecated and removed since react-native 0.47, if you modify this file (/Users/hieunguyen/Mobile/ArduinoBT/node_modules/react-native-bluetooth-serial/android/src/main/ja va/com/rusel/RCTBluetoothSerial/RCTBluetoothSerialPackage.java) and simply remove the @Override just before the createJSModules method, that should fix it temporarily.

Your other choice is to use an older version of RN.

ithieund, 2018-02-05:
@darkrift ,
Thank you very much, I tried and it worked, but some others issue occurred :(
Do you know about the roadmap for this project? Is it stoped maintaining?

Thanks.

darkrift, 2018-02-05:
@ithieund I have no idea of the roadmap of this project. I was trying to find a library doing bluetooth classic and it seems we are pretty limited in terms of plugins for that purpose (found only 1 other than this one). While trying this one, I had that issue and fixed it temporarily so I could actually give it a better try.

esbozos, 2018-05-24:
Just testing this library, this error is easy to solve. Just remove @ Override  from /node_modules/react-native-bluetooth-serial/android/src/main/java/com/rusel/RCTBluetoothSerial/RCTBluetoothSerialPackage.java

from this

`public class RCTBluetoothSerialPackage implements ReactPackage {
    static final String TAG = "BluetoothSerial";

    @Override
    public List<NativeModule> createNativeModules(ReactApplicationContext reactContext) {
        List<NativeModule> modules = new ArrayList<>();
        modules.add(new RCTBluetoothSerialModule(reactContext));
        return modules;
    }

    @Override
    public List<Class<? extends JavaScriptModule>> createJSModules() {
        return Collections.emptyList();
    }

    @Override
    public List<ViewManager> createViewManagers(ReactApplicationContext reactContext) {
        return Collections.emptyList();
    }
}`
public class RCTBluetoothSerialPackage implements ReactPackage {
    static final String TAG = "BluetoothSerial";

    @Override
    public List<NativeModule> createNativeModules(ReactApplicationContext reactContext) {
        List<NativeModule> modules = new ArrayList<>();
        modules.add(new RCTBluetoothSerialModule(reactContext));
        return modules;
    }

    @Override
    public List<Class<? extends JavaScriptModule>> createJSModules() {
        return Collections.emptyList();
    }

    @Override
    public List<ViewManager> createViewManagers(ReactApplicationContext reactContext) {
        return Collections.emptyList();
    }
}

to this
`public class RCTBluetoothSerialPackage implements ReactPackage {
    static final String TAG = "BluetoothSerial";

    
    public List<NativeModule> createNativeModules(ReactApplicationContext reactContext) {
        List<NativeModule> modules = new ArrayList<>();
        modules.add(new RCTBluetoothSerialModule(reactContext));
        return modules;
    }

    
    public List<Class<? extends JavaScriptModule>> createJSModules() {
        return Collections.emptyList();
    }

    
    public List<ViewManager> createViewManagers(ReactApplicationContext reactContext) {
        return Collections.emptyList();
    }
}`

ithieund, 2018-05-24:
@esbozos ,
Thank you very much. I can make it work already.

It seems that the author of this plugins does not maintain it any more?
How about forking this into another repository and imporove it?

Can you do that?
Thanks.

WhaSukGO, 2020-03-21:
For anyone using react-native 0.61.5, you need to disable autolinking for this package.

https://github.com/react-native-community/cli/blob/master/docs/autolinking.md

After removing @Override and disabling autolinking, it works!

whatdtech, 2020-11-24:
@WhaSukGO  Thank you man. It worked