rusel1989/react-native-bluetooth-serial #71: How to send Hex Values via Bluetooth

vladinko0, 2018-02-16:
In the BluetoothSerialExample I see:

```
 write (message) {
    BluetoothSerial.write(message)
    .then((res) => {
      Toast.showShortBottom('Successfuly wrote to device')
      this.setState({ connected: true })
    })
    .catch((err) => Toast.showShortBottom(err.message))
  }
```

But I need to send the Enquiry ENQ (0x05). And if I store into the "message" value 0x05, slave device gives me no response, because it probably wants hex value 0x05, not string value.

In the ..\react-native-bluetooth-serial-master\index.js I see:  

```
BluetoothSerial.write = (data) => {
  if (typeof data === 'string') {
    data = new Buffer(data)
  }
  return BluetoothSerial.writeToDevice(data.toString('base64'))
}  
```
Should I replace data.toString('base64') with data('base64') to pass hex value, or should I in ..\node_modules\react-native-bluetooth-serial\android\src\main\java\com\rusel\RCTBluetoothSerial\RCTBluetoothSerialModule.java:

```
 public void writeToDevice(String message, Promise promise) {
        if (D) Log.d(TAG, "Write " + message);
        byte[] data = Base64.decode(message, Base64.DEFAULT);
        mBluetoothService.write(data);
        promise.resolve(true);
    }
```
replace String message with Int message in order to send hex value 0x05?

Please help me. I feel so lost.

mshavandi, 2018-05-28:
I also need to pass hex value with this package. Can anyone help please?

PiotrDabek, 2018-07-31:
Have you managed to solve this problem? I'm facing similar one right now. Any help would be appreciated

JackTam1993, 2018-10-23:
Here is my solution:
```
let buf = new Buffer("AA22EC00000000000102AA5500023C4E000001007000000000002300000000000000D0", "hex");
BluetoothSerial.write(buf)
```
Hope that still helpful.

chamaloriz, 2019-09-29:
### I found how to :p  spent 2 Days
basically the array wasn't showing as a uint8array but as a typedArray 
Buffer.from() fixed it

    let encoder = new EscPosEncoder();

    let result = encoder
      .initialize()
      .text("HelloWorld")
      .newline()
      .qrcode('HelloWorld')
      .encode();

    const { Buffer } = require('buffer');

    BluetoothSerial.write(Buffer.from(result.buffer))