rusel1989/react-native-bluetooth-serial #73: How I Print text in bold

mtahir08, 2018-03-10:
I want to print some values in bold. 
Can anyone knows how I can do this.
Thanks

VyankeshH, 2018-04-05:
@mtahir08 did you got a solution for this

mtahir08, 2018-04-05:
@VyankeshH I did it using escpos-xml package.
https://www.npmjs.com/package/escpos-xml

But now I am facing issue in auto cut paper.

VyankeshH, 2018-04-06:
undefined is not a function(evaluating '_escposXml.EscPos.getBufferXML(xml)')
I got this type of error when I am trying with this library.Do you have any solution for this??

VyankeshH, 2018-04-06:
@mtahir08 did you got qrcode using this.

mtahir08, 2018-04-06:
```
import { EscPos } from 'escpos-xml';
 const xml = `
    <?xml version="1.0" encoding="UTF-8"?>
    <document>
        <line-feed />
        <align mode="center">
            <bold>
                <text-line size="1:0">{{name}}</text-line>
            </bold>
            <line-feed />
            <small>
                <text-line size="1:0">---------------------</text-line>
                <text-line size="1:0">Dine-Inn</text-line>
                <text-line size="1:0">---------------------</text-line>
            </small>
            <line-feed />
            <text-line>Date & Time: {{moment date format="DD/MM/YYYY HH:MM"}} </text-line>
        </align>
          <line-feed />
    </document>
  `      
    const data = {
       name : 'Tahir' ,
      date:new Date()
    }

    const buffer_data = EscPos.getBufferFromTemplate(xml, data);
```

@VyankeshH Here is sample code. 
No I yet not use qrcode.

afrakhan123, 2018-07-19:
I did it like this (basically wrote my own render function):

   data.map((k, i) => {
      let row = k.x + "                                           ";
      row =
        row.slice(0, 28 - k.y.toString().length) +
        k.y +
        "                            ";

      row =
        row.slice(0, 40 - Math.round(k.amount).toString().length) +
        Math.round(k.amount) +
        "\n";
      tableText += row;
    });

afrakhan123, 2018-07-19:
[image]