rusel1989/react-native-bluetooth-serial #77: Unable to reconnect to device if connection lost...

ZombienDK, 2018-04-09:
Hello...

I am wondering how i would be able to reconnect to an device if connection is lost for some reason!

i have tried a simple interval, based on the isConnected(), where i grab the last connected the device information like mac adress / uid and so on and try to connect if it returns false...

setInterval(() => {
      BluetoothSerial.isConnected().then((res) => {
        if (!res) {

          getDevice().then((device) => {

            if ( device.id !== undefined ) {

              console.log(device)

              BluetoothSerial.connect(device.id).then((res) => {

                console.log('Test')
                
                //console.log(`Connected to device ${device.name}`)
              }).catch((err) => { })

            }

          })

        } 
      })   
    },5000)

but i run into a crash in the app after sometime...

( From the log )
04-09 09:15:05.168 25784 25933 W ReactNativeJS: Possible Unhandled Promise Rejection (id: 7):
04-09 09:15:05.168 25784 25933 W ReactNativeJS: Unable to connect to device
04-09 09:15:05.168 25784 25933 W ReactNativeJS: createErrorFromErrorData@http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=false&minify=false:4712:24
04-09 09:15:05.168 25784 25933 W ReactNativeJS: http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=false&minify=false:4670:49
04-09 09:15:05.168 25784 25933 W ReactNativeJS: __invokeCallback@http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=false&minify=false:4985:21
04-09 09:15:05.168 25784 25933 W ReactNativeJS: http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=false&minify=false:4844:32
04-09 09:15:05.168 25784 25933 W ReactNativeJS: guard@http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=false&minify=false:4783:7
04-09 09:15:05.168 25784 25933 W ReactNativeJS: invokeCallbackAndReturnFlushedQueue@http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=false&minify=false:4843:12
04-09 09:15:05.168 25784 25933 W ReactNativeJS: invokeCallbackAndReturnFlushedQueue@[native code]

04-09 09:37:29.228 29312 29387 E ReactNativeJS: The callback connect() exists in module BluetoothSerial, but only one callback may be registered to a function in a native module.
04-09 09:37:29.537 29312 29387 W ReactNativeJS: Unable to symbolicate stack trace: The stack is null

tanmaybhatt, 2018-07-04:
Having same issue, did you find any solution?

freemanUA, 2019-03-20:
Got the same issue by using timeout. Ended with modal for user with button to reconnect or cancel.

ZombienDK, 2019-05-20:
Sorry for the late response…

Well i got it working so my app where able to reconnect.
The issue where that i tried to multiple request to a single device, before the lib could register a response or timeout in the request. And that where the reason why it crashed.