rusel1989/react-native-bluetooth-serial #85: Remove paired device

sadidebbal-rhizom-team, 2018-07-17:
Hello, i'm using your library to communicate with differents devices in bluetooth, I've a paired device list and would like to remove a device from the list if i choose to delete it. 
It is possible ? there is a method in the library ?

arasabbasi, 2018-10-04:
You can not unpair a bluetooth device per native app. You have to do it in the settings.

caiorios, 2019-12-25:
Hello, I would like to know why it isn't possible. There is a function called `unpairDevice` that apparently does nothing.

When I try to call it, the log stops on `Start Unpairing...`. I suspect onReceive method is not been called. However, I don't know Android stuff.

https://github.com/rusel1989/react-native-bluetooth-serial/blob/09434a0726697902f52ea3a58790fc19f6f14dbe/android/src/main/java/com/rusel/RCTBluetoothSerial/RCTBluetoothSerialModule.java#L596