rusel1989/react-native-bluetooth-serial #89: Can not run correctly

LiuduZero, 2018-08-23:
I operate follow the README.md,  but  when i   run 'react-native run-android' , the terminal shows correctly, but my cellphone show the red screen and  'Could not get BatchedBridge, make sure your bundle is packaged correctly' on it .  So  what should I do?

jhonber, 2018-08-24:
It seems that this repository is no longer maintained, I have make a fork and fix some errors.
You can run and test here https://github.com/jhonber/react-native-bluetooth-serial

jeffreyvanhees, 2018-09-17:
> It seems that this repository is no longer maintained, I have made a fork and fixed some errors.
> You can run and test it here https://github.com/jhonber/react-native-bluetooth-serial

You're a hero!