rusel1989/react-native-bluetooth-serial #94: WARNING: Configuration 'compile' is obsolete and has been replaced with 'implementation' and 'api'. It will be removed at the end of 2018. For more information see: http://d.android.com/r/tools/update-dependency-configurations.html

Shhzdmrz, 2018-10-09:
Getting multiple warnings on version **1.0.0-rc1** any plan to fix before end of the year?

> WARNING: Configuration 'compile' is obsolete and has been replaced with 'implementation' and 'api'.
> It will be removed at the end of 2018. For more information see: http://d.android.com/r/tools/update-dependency-configurations.html
> 

> WARNING: The specified Android SDK Build Tools version (23.0.1) is ignored, as it is below the minimum supported version (27.0.3) for Android Gradle Plugin 3.1.4.
> Android SDK Build Tools 27.0.3 will be used.

> To suppress this warning, remove "buildToolsVersion '23.0.1'" from your build.gradle file, as each version of the Android Gradle Plugin now has a default version of the build tools.

bkitduy, 2018-10-09:
It's not an issue of this project. Replace 'compile' by 'implementation' and upgrade Android SDK Build Tools to hide this warning

Shhzdmrz, 2018-10-09:
@bkitduy Are you sure? Every time I use yarn or npm command I have to replace it. It's not a good approach.

closed 2020-06-12 by Shhzdmrz