rusel1989/react-native-bluetooth-serial #97: Build Failure - Methods

mthrsj, 2018-12-18:
I've been trying to build many times and received the error below:

> Task :react-native-bluetooth-serial:compileDebugJavaWithJavac FAILED
C:\Users\mathe\Documents\GitHub\natuino\node_modules\react-native-bluetooth-serial\android\src\main\java\com\rusel\RCTBluetoothSerial\RCTBluetoothSerialPackage.java:23: error: method does not override or implement a method from a supertype
    @Override
    ^
1 error
FAILURE: Build failed with an exception.

My dependecies: 

```
"dependencies": {
    "react": "^16.5.1",
    "react-native": "^0.57.3",
    "react-native-bluetooth-serial": "^1.0.0-rc1",
    "react-native-elements": "^0.19.1",
    "react-native-vector-icons": "^6.1.0",
    "react-navigation": "^2.18.2",
    "react-redux": "^6.0.0",
    "redux": "^4.0.1"
}
```

The solution was to open the `RCTBluetoothSerialPackage.java` and remove the `@Override`. Please, fix it, if possible.

1rjun, 2018-12-29:
Just remoe the @overiride from base java class

mthrsj, 2018-12-29:
@1rjun, I did. But I shouldn't need to do this to get the package working.

jetobe95, 2019-03-29:
> Just remoe the @overiride from base java class

Thanks Works
com/rusel/RCTBluetoothSerial/RCTBluetoothSerialPackage.java:23

amrudesh-santhanam, 2020-03-05:
Removing the @override works.

mcavalcante28, 2020-03-31:
Remove @override from line number 23 works for me.