rusel1989/react-native-bluetooth-serial #99: Activity triggering using react-native-bluetooth-serial

mabn123, 2019-02-03:
Is it possible to trigger a navigation to a different activity/screen after detecting a specified bluetooth id? and how? I have been trying to make it happen but weren't successful

ivankfit, 2019-03-26:
this should give you a hint
connect (device) {
    this.setState({ connecting: true })
    BluetoothSerial.connect(device.id)
    .then((res) => {
      console.log(`Connected to device ${device.name}`)
      this.setState({ device, connected: true, connecting: false })
//////add this line to the connect function
      this.props.navigation.navigate('Second',{device:device});
    })
    .catch((err) => console.log(err.message))
  }

mabn123, 2019-03-28:
Thank you very much for the hint, however, my case requires me to utilize the BluetoothSerial.discoverUnpairedDevices() to scan for available unpaired devices and when a determined bluetooth id say <AA:BB:CC:DD:EE:FF> is scanned, the function would automatically trigger a different activity/screen

mabn123, 2019-03-28:
This is what I have come out so far but it didn't work as the activity was never triggered:
`      this.setState({ discovering: true })
      BluetoothSerial.discoverUnpairedDevices()
      .then((unpairedDevices) => {
        this.setState({ unpairedDevices, discovering: false })
      })
      .catch((err) => Toast.showShortBottom(err.message))

        unpairedDevices.filter(function(device){
          return device.id == this.state.DEVICE_ID;
        }).map(function(device,i){
            if (device.id === this.state.DEVICE_ID){
              this.props.navigation.navigate('QRScan',{
                STUDENT_ID: Student_ID,
                PROG_ID: PROG_ID,
                DEVICE_ID : Device_ID,
                COURSE_NAME : Course_Name,
                COURSE_ID : this.props.navigation.state.params.CO_ID,
                latitude : this.state.latitude,
                longitude : this.state.longitude,
              });
            }
        });`

mabn123, 2019-03-28:
This is what I have come out so far but it didn't work as the activity was never triggered:

`  discoverUnpaired () {

    const { unpairedDevices } = this.state ;

    if (this.state.discovering) {
      return false
    } 

    else {
      this.setState({ discovering: true })
      BluetoothSerial.discoverUnpairedDevices()
      .then((unpairedDevices) => {
        this.setState({ unpairedDevices, discovering: false })
      })
      .catch((err) => Toast.showShortBottom(err.message))

        unpairedDevices.filter(function(device){
          return device.id == this.state.DEVICE_ID;
        }).map(function(device,i){
            if (device.id === this.state.DEVICE_ID){
              this.props.navigation.navigate('QRScan',{
                STUDENT_ID: Student_ID,
                PROG_ID: PROG_ID,
                DEVICE_ID : Device_ID,
                COURSE_NAME : Course_Name,
                COURSE_ID : this.props.navigation.state.params.CO_ID,
                latitude : this.state.latitude,
                longitude : this.state.longitude,
              });
            }
        });

    }
  }`

mabn123, 2019-03-28:
This is what I have come out so far but it didn't work as the activity was never triggered:
`
  discoverUnpaired () {

    const { unpairedDevices } = this.state ;

    if (this.state.discovering) {
      return false
    } 

    else {
      this.setState({ discovering: true })
      BluetoothSerial.discoverUnpairedDevices()
      .then((unpairedDevices) => {
        this.setState({ unpairedDevices, discovering: false })
      })
      .catch((err) => Toast.showShortBottom(err.message))

        unpairedDevices.filter(function(device){
          return device.id == this.state.DEVICE_ID;
        }).map(function(device,i){
            if (device.id === this.state.DEVICE_ID){
              this.props.navigation.navigate('QRScan',{
                STUDENT_ID: Student_ID,
                PROG_ID: PROG_ID,
                DEVICE_ID : Device_ID,
                COURSE_NAME : Course_Name,
                COURSE_ID : this.props.navigation.state.params.CO_ID,
                latitude : this.state.latitude,
                longitude : this.state.longitude,
              });
            }
        });

    }
  }
`

mabn123, 2019-03-28:
This is what I have come out so far but the activity never triggered despite the fact that the bluetooth signal exist:
`
  discoverUnpaired () {

    const { unpairedDevices } = this.state ;

    if (this.state.discovering) {
      return false
    } 

    else {
      this.setState({ discovering: true })
      BluetoothSerial.discoverUnpairedDevices()
      .then((unpairedDevices) => {
        this.setState({ unpairedDevices, discovering: false })
      })
      .catch((err) => Toast.showShortBottom(err.message))

        unpairedDevices.filter(function(device){
          return device.id == this.state.DEVICE_ID;
        }).map(function(device,i){
            if (device.id === this.state.DEVICE_ID){
              this.props.navigation.navigate('Second');
            }
        });

    }
  }

`

mabn123, 2019-03-28:
The 'this.state.DEVICE_ID' is where the target device's device id is kept