rusel1989/react-native-bluetooth-serial #100: Can't write and disconnect.

manelsanz, 2019-02-14:
Hi, I can connect perfectly to my bluetooth serial port, but when I call `BluetoothSerial.connect()`, it shows the following error: `RTCBluetoothSerial.writeToDevice got 4 arguments, expected 3`.

The same with the "disconnect" method: `RTCBluetoothSerial.writeToDevice got 3 arguments, expected 2`.

I need some help, please...

chamaloriz, 2019-09-29:
### I found how to :p  spent 2 Days

    let encoder = new EscPosEncoder();

    let result = encoder
      .initialize()
      .text("HelloWorld")
      .newline()
      .qrcode('HelloWorld')
      .encode();

    const { Buffer } = require('buffer');

    BluetoothSerial.write(Buffer.from(result.buffer))

tienw, 2019-11-06:
I had the same issue and your solution works! However, I recieved the following error after printed

`Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference `

Anyone have same issue? it looks like a Java issue