rusel1989/react-native-bluetooth-serial #102: Execution failed for task ':react-native-bluetooth-serial:verifyReleaseResources'.

CaReasts, 2019-04-03:
java.util.concurrent.ExecutionException: com.android.builder.internal.aapt.v2.Aapt2Exception: Android resource linking failed
  Output:  error: resource android:style/TextAppearance.Material.Widget.Button.Borderless.Colored not found.
  error: resource android:style/TextAppearance.Material.Widget.Button.Colored not found.
  /Users/huidongche/Documents/sz/workflow/workflow-mobile/node_modules/react-native-bluetooth-serial/android/build/intermediates/res/merged/release/values-v26/values-v26.xml:7: error: resource android:attr/colorError not found.
  /Users/huidongche/Documents/sz/workflow/workflow-mobile/node_modules/react-native-bluetooth-serial/android/build/intermediates/res/merged/release/values-v26/values-v26.xml:11: error: resource android:attr/colorError not found.
  /Users/huidongche/Documents/sz/workflow/workflow-mobile/node_modules/react-native-bluetooth-serial/android/build/intermediates/res/merged/release/values-v26/values-v26.xml:15: error: style attribute 'android:attr/keyboardNavigationCluster' not found.
  /Users/huidongche/Documents/sz/workflow/workflow-mobile/node_modules/react-native-bluetooth-serial/android/build/intermediates/res/merged/release/values/values.xml:226: error: resource android:attr/fontStyle not found.
  /Users/huidongche/Documents/sz/workflow/workflow-mobile/node_modules/react-native-bluetooth-serial/android/build/intermediates/res/merged/release/values/values.xml:226: error: resource android:attr/font not found.
  /Users/huidongche/Documents/sz/workflow/workflow-mobile/node_modules/react-native-bluetooth-serial/android/build/intermediates/res/merged/release/values/values.xml:226: error: resource android:attr/fontWeight not found.
  error: failed linking references.
  
  Command: /Users/huidongche/.gradle/caches/transforms-1/files-1.1/aapt2-3.2.0-4818971-osx.jar/35e5db7c76f122562c5cd109f87be003/aapt2-3.2.0-4818971-osx/aapt2 link -I\

 ext {
        buildToolsVersion = "27.0.3"
        minSdkVersion = 19
        compileSdkVersion = 27
        targetSdkVersion = 26
        supportLibVersion = "27.1.1"
    }

pawel2210, 2019-04-11:
Got the same problem. Did you resolve it anyway?

CaReasts, 2019-04-12:
> 得到了同样的问题。你有没有解决它？

no

CaReasts, 2019-04-12:
> Got the same problem. Did you resolve it anyway?

no

pawel2210, 2019-04-12:
I finally did. This solution helped me:
https://github.com/facebook/react-native/issues/19239#issuecomment-439232501
I have also installed this libs before:
libncurses5:i386 libstdc++6:i386 zlib1g:i386 (suggested somewhere in linked topic)
but without an effect, so I suppose that adding to android/build.gradle subprojects code was the solution.

hariom1995, 2020-09-30:
> Task :react-native-bluetooth-serial:verifyReleaseResources FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-bluetooth-serial:verifyReleaseResources'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
   > Android resource linking failed
     /home/hari/.gradle/caches/transforms-2/files-2.1/37631cd0ab1b46aeb23414622984ad25/appcompat-1.0.2/res/values-v26/values-v26.xml:5:5-8:13: AAPT: error: resource android:attr/colorError not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/37631cd0ab1b46aeb23414622984ad25/appcompat-1.0.2/res/values-v26/values-v26.xml:9:5-12:13: AAPT: error: resource android:attr/colorError not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/37631cd0ab1b46aeb23414622984ad25/appcompat-1.0.2/res/values-v26/values-v26.xml:13:5-16:13: AAPT: error: style attribute 'android:attr/keyboardNavigationCluster' not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/37631cd0ab1b46aeb23414622984ad25/appcompat-1.0.2/res/values-v28/values-v28.xml:5:5-8:13: AAPT: error: resource android:attr/dialogCornerRadius not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/37631cd0ab1b46aeb23414622984ad25/appcompat-1.0.2/res/values-v28/values-v28.xml:9:5-12:13: AAPT: error: resource android:attr/dialogCornerRadius not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/1cd17b4a4c7e4b94a301254fec05bdc3/core-1.0.1/res/values/values.xml:57:5-88:25: AAPT: error: resource android:attr/fontStyle not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/1cd17b4a4c7e4b94a301254fec05bdc3/core-1.0.1/res/values/values.xml:57:5-88:25: AAPT: error: resource android:attr/font not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/1cd17b4a4c7e4b94a301254fec05bdc3/core-1.0.1/res/values/values.xml:57:5-88:25: AAPT: error: resource android:attr/fontWeight not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/1cd17b4a4c7e4b94a301254fec05bdc3/core-1.0.1/res/values/values.xml:57:5-88:25: AAPT: error: resource android:attr/fontVariationSettings not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/1cd17b4a4c7e4b94a301254fec05bdc3/core-1.0.1/res/values/values.xml:57:5-88:25: AAPT: error: resource android:attr/ttcIndex not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/1cd17b4a4c7e4b94a301254fec05bdc3/core-1.0.1/res/values/values.xml:89:5-125:25: AAPT: error: resource android:attr/startX not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/1cd17b4a4c7e4b94a301254fec05bdc3/core-1.0.1/res/values/values.xml:89:5-125:25: AAPT: error: resource android:attr/startY not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/1cd17b4a4c7e4b94a301254fec05bdc3/core-1.0.1/res/values/values.xml:89:5-125:25: AAPT: error: resource android:attr/endX not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/1cd17b4a4c7e4b94a301254fec05bdc3/core-1.0.1/res/values/values.xml:89:5-125:25: AAPT: error: resource android:attr/endY not found.
         
     /home/hari/.gradle/caches/transforms-2/files-2.1/1cd17b4a4c7e4b94a301254fec05bdc3/core-1.0.1/res/values/values.xml:126:5-132:25: AAPT: error: resource android:attr/offset not found.
         

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.2/userguide/command_line_interface.html#sec:command_line_warnings

buildscript {
    ext {
        buildToolsVersion = "29.0.2"
        minSdkVersion = 28
        compileSdkVersion = 29
        targetSdkVersion = 29
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:3.5.3")
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

Facing the same issue..

Viral-Gajera, 2023-11-03:
### I was having the same issue 
>  Task :react-native-bluetooth-serial:verifyReleaseResources FAILED

with this project configuration 

[image]

But [this](https://github.com/facebook/react-native/issues/19239#issuecomment-502784768) work for me...