rusel1989/react-native-bluetooth-serial #107: null is not an object evaluating Bluetooth.on

mamdouh97, 2019-08-27:
Basically, I created a blank react-native project and installed the library and did the installation manually. when I import the library in App.js without using it the application throws an error.
null is not an object evaluating Bluetooth.on

mrzephros, 2019-08-29:
Same issue...

hszczapa, 2019-11-08:
Has this been solved? I'm having the same issue.

bhoomika911, 2020-02-25:
I am also facing the same issue in iOS simulator. Anyone can please help here?

lucassouza16, 2021-02-21:
I have been using this module for two of my projects for a long time, but in recent versions of react it presented several problems, mainly in iOS, I looked for other modules that work with esc pos and in my opinion this is the most complete and practical, because unfortunately the project was abandoned. So I thought it best to fix the module problems to continue using it, for those who have problems feel free to use:

https://github.com/lucassouza16/react-native-bluetooth-serial.git

To replace the original with the fork:

iOS

yarn remove react-native-bluetooth-serial && yarn add lucassouza16/react-native-bluetooth-serial#latest && cd ios && pod install && cd .. && yarn ios

Android

yarn remove react-native-bluetooth-serial && yarn add lucassouza16/react-native-bluetooth-serial#latest && cd android && ./gradlew clean && cd .. && yarn android

bharathram89, 2021-02-23:
@lucassouza16 you are the real hero. using ur repo now :)

sandeepsandy8790, 2021-03-05:
Thanks @lucassouza16 for the solution

Rinrin56, 2021-03-21:
@lucassouza16  hi  when i run this "yarn add lucassouza16/react-native-bluetooth-serial#latest"
something error:
error An unexpected error occurred: "https://raw.githubusercontent.com/lucassouza16/react-native-bluetooth-serial/b8060e718dc71ab50a89aadc28fca8d34d50dabe/package.json: getaddrinfo ENOENT raw.githubusercontent.com".

i don't what happen

lucassouza16, 2021-03-21:
> @lucassouza16 hi when i run this "yarn add lucassouza16/react-native-bluetooth-serial#latest"
> something error:
> error An unexpected error occurred: [...]

This error is still very vague, are you sure that only this is returned?