rusel1989/react-native-bluetooth-serial #108: Error IOS

thinh2408, 2019-08-28:
'React/RCTBridgeModule.h' file not found

arevaldez, 2019-09-23:
This is not an issue with the library, this is a React issue. It could be many things. It could be a PodFile issue, a Pods and React modules running in parallel issue, etc. But, this isn't an issue with the library itself. If you can, I'd try running it on just Android. It'll probably be okay!

datvp09, 2019-10-15:
@arevaldez how can I fix it?
My project use react-native 0.61 which doesn't have React podfile

alex-valdez, 2019-10-15:
@datvp09 you have to install the pod files. Typically by going into the iOS folder and running `pod install`

arevaldez, 2019-10-15:
@datvp09 you have to install the pod files. Typically by going into the iOS folder and running pod install

nvdnvd00, 2019-11-05:
I have the same issue, any idea?

afrakhan123, 2019-12-29:
I have the same issue on react native 0.61, i got it to work with older react native versions

patroldo, 2020-02-28:
I figured out how to resolve the issue. Btw, here is how to make this repository available to install via pods.
2)
In the folder ```{YOUR_PROJECT}/node_modules/react-native-bluetooth-serial``` add file
`react-native-bluetooth-serial.podspec`

3) Insert into newly created file follow content:
```
require 'json'

package = JSON.parse(File.read(File.join(__dir__, 'package.json')))

Pod::Spec.new do |s|
  s.name         = package['name']
  s.version      = package['version']
  s.summary      = package['description']
  s.license      = package['license']

  s.authors      = package['author']
  s.homepage     = package['homepage']
  s.platform     = :ios, "9.0"

  s.source       = { :git => "https://github.com/rusel1989/react-native-bluetooth-serial.git", :tag => "v#{s.version}" }
  s.source_files  = "ios/**/*.{h,m}"

  s.dependency 'React'
end
```
4) Fix file `{YOUR_PROJECT}/node_modules/react-native-bluetooth-serial/ios/RCTBluetoothSerial/RCTBluetoothSerial.h`, above line

```#import <React/RCTBridgeModule.h>```

add follow import
```#import <React/RCTBridge.h>```

5) Make **clean build folder**

6) In root of your project enter ios folder and make
```pod install```

If all okay, the project should finally builds. And you should get troubled with #107. But 

@rusel1989 Could you approve above changes to repository please? At least add pod file?

WhaSukGO, 2020-03-27:
> I have the same issue on react native 0.61, i got it to work with older react native versions

@afrakhan123 May I ask which version of React-native you're using?

lchenfox, 2020-08-13:
The same issue here with `React Native 0.63.2`. Any solution?

keselme90, 2021-04-26:
@patroldo I tried your solution, and now I get `'React/RCTBridge.h' file not found` error when I try to build.
Do you know why it might happen?

keselme90, 2021-05-04:
@lchenfox did you find a solution?