rusel1989/react-native-bluetooth-serial #112: How to read data?

GPXNB, 2019-11-08:
Use of read?
How to get Bluetooth information?

JulioCVaz, 2019-12-03:
You can use the method BluetoothSerial.readFromDevice() to read data sent by device. Use a setInterval and execute or you can add a event listener DEVICE_READ, consult the file RTCBluetoothSerialModule.java on node_modules

GHOSTLORE, 2019-12-04:
> You can use the method BluetoothSerial.readFromDevice() to read data sent by device. Use a setInterval and execute or you can add a event listener DEVICE_READ, consult the file RTCBluetoothSerialModule.java on node_modules

can you show me a runnable case?Thank you so much

JulioCVaz, 2019-12-05:
Sure, for exemple:

import BluetoothSerial from 'react-native-bluetooth-serial';
import { Buffer } from 'buffer';
global.Buffer = Buffer;

receiveDataFromDevice = async () => {
    const response = await BluetoothSerial.readFromDevice();
    return response;
  }

and in your .js file execute a setInterval repeatedly.

notes: The library BluetoothSerial contains some methods in your interface React, but the method **readFromDevice** is in the RCT .java file on node_modules / react-native-bluetooth-serial ...

GHOSTLORE, 2019-12-05:
> Sure, for exemple:
>
> import BluetoothSerial from 'react-native-bluetooth-serial';
> import { Buffer } from 'buffer';
> global.Buffer = Buffer;
>
> receiveDataFromDevice = async () => {
> const response = [...]

Thank you so much!But where should I place this code...? setInterval call what?Can you show me more code ,I was stuck in this problem for a long time

ramacha7, 2021-02-25:
Hi, @JulioCVaz  I am trying to read data from a microcontroller that is sending data via a Bluetooth module (HC05). I see that it has been said that we can use the BluetoothSerial.readFromDevice(). I was wondering how I could actually write implementable code in my .js file using the event listener?  Also, I wanted to ask if we can use all the functions that are available in the file RTCBluetoothSerialModule.java located in node_modules?

JulioCVaz, 2021-02-25:
@ramacha7 , Hello friend!

you can see in this file, methods that have a "@ReactMethod" you can implements in the your .js code.

https://github.com/rusel1989/react-native-bluetooth-serial/blob/master/android/src/main/java/com/rusel/RCTBluetoothSerial/RCTBluetoothSerialModule.java

And in this file you can see that method to use a ListenerEvent

https://github.com/rusel1989/react-native-bluetooth-serial/blob/master/index.js

Example for implementation follow this file:
https://github.com/rusel1989/react-native-bluetooth-serial/blob/master/BluetoothSerialExample/src/App.js

good luck :)

ramacha7, 2021-02-25:
@JulioCVaz Thank you so much for the direction. I also wanted to ask if I can directly use the readFromDevice() function, because I need to use '\r' as my delimiter and I don't think it uses that as the delimiter.

JulioCVaz, 2021-02-25:
@ramacha7 Yeah, you can! Maybe you can need adjust the data received using \r

ramacha7, 2021-02-25:
@JulioCVaz I tried setting up an event and using the readFromDevice() function. However I am not able to read the data being sent. This is my code so far.
`          BluetoothSerial.withDelimiter('\r\n').then((res) => {
              console.log("Delimiter has been set up ");
              BluetoothSerial.on('read',handleRead);  
          })

    const handleRead = () => {
        console.log("entered read function");
        const result = BluetoothSerial.readFromDevice();
        setreaddata(result);
        console.log(readdata);
    }`

JulioCVaz, 2021-02-25:
@ramacha7 You should use async await cuz this method readFromDevice return a promise.

example:

const handleRead = async () => {
    console.log("entered read function");
    const result = await BluetoothSerial.readFromDevice();
    setreaddata(result);
    console.log(readdata);
}`

ramacha7, 2021-02-25:
@JulioCVaz . I tried the code you suggested and I was still not able to receive the data. the event is being triggered however, I cant see the data. This is the output:

[Thu Feb 25 2021 14:51:09.913]  LOG      Permission is OK
[Thu Feb 25 2021 14:51:09.914]  LOG      Delimiter has been set up 
[Thu Feb 25 2021 14:51:42.202]  LOG      entered read function
[Thu Feb 25 2021 14:51:42.210]  LOG      entered read function
[Thu Feb 25 2021 14:51:42.218]  LOG      entered read function
[Thu Feb 25 2021 14:51:42.224]  LOG

JulioCVaz, 2021-02-25:
@ramacha7 you pair the device with your cellphone? you only can read the data only if the devices is paired.

ramacha7, 2021-02-25:
@JulioCVaz Oh are you talking about the function pairdevice()? Are you referring to this function? I pair the device with my cellphone in the settings of my android. But I don't write any code to do it.

    private void pairDevice(BluetoothDevice device) {
        try {
            if (D) Log.d(TAG, "Start Pairing...");
            Method m = device.getClass().getMethod("createBond", (Class[]) null);
            m.invoke(device, (Object[]) null);
            registerDevicePairingReceiver(device.getAddress(), BluetoothDevice.BOND_BONDED);
            if (D) Log.d(TAG, "Pairing finished.");
        } catch (Exception e) {
            Log.e(TAG, "Cannot pair device", e);
            if (mPairDevicePromise != null) {
                mPairDevicePromise.reject(e);
                mPairDevicePromise = null;
            }
            onError(e);
        }
    }

JulioCVaz, 2021-02-25:
@ramacha7 I think that how you pair into android settings should work. But I would try pair using pairDevice() check if has another behavior.

taj567, 2022-01-06:
@JulioCVaz hello, am able to pair the device but not able receive any data from it using BluetoothSerial.readFromDevice().
  readFromDevice = async id => {
    console.log('entered read function');
    const result = await BluetoothSerial.readFromDevice(id).then(device => {
      console.log(device, 'device');
    });
    console.log(result, 'entered read function');
  };

result --  entered read function
 LOG  {"_U": 0, "_V": 0, "_W": null, "_X": null} readFromDevice
 LOG  {"_U": 0, "_V": 0, "_W": null, "_X": null} readOnce

JulioCVaz, 2022-01-06:
Hi @taj567 , how are u?
Did you try to use async wait without o `.then()` ?

Maybe that way it will work

```
 readFromDevice = async (id) => {
  const res = await BluetoothSerial.readFromDevice(id)
  console.log(res)
 }
```

taj567, 2022-01-07:
Hi @JulioCVaz, am fine & Hope you to?
from the past 10 days working on Philips CPAP machine in react native Bluetooth worked on three libraries 
NOTE: need to retrieve the data after pairing the CPAP machine through bluetooth.
1,react-native-ble 
2,react-native-serial-next
3,react-native-ble-manager 
as of now from ble-manager, we are able to receive some data but related information was not retrieved from the device.

able to retrieve----
 {"advertising": {"isConnectable": true, "localName": "PR BT FC6F", 
"manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "AgEACwlQUiBCVCBGQzZGB/8AAAAAAAARB/NZ9KGQ4fyu9EkJ+H+GU0UAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {},
 "serviceUUIDs": ["4553867f-f809-49f4-aefc-e190a1f459f3"], "txPowerLevel": -2147483648}, "characteristics": [{"characteristic": "2a05", "properties": [Object], "service": "1801"},
 {"characteristic": "22a4e311-a097-4517-9b81-cf32af60b982", "descriptors": [Array], "properties": [Object], "service": "4553867f-f809-49f4-aefc-e190a1f459f3"}],
 "id": "00:1F:FF:A7:6D:63", "name": "PR BT FC6F", "rssi": -39, "services": [{"uuid": "1801"}, {"uuid": "4553867f-f809-49f4-aefc-e190a1f459f3"}]}

-----------"bytes": [2, 1, 0, 11, 9, 80, 82, 32, 66, 84, 32, 70, 67, 54, 70, 7, 255, 0, 0, 0, 0, 0, 0, 17, 7, 243, 89, 244, 161, 144, 225, 252, 174, 244, 73, 9, 248, 127, 134, 83, 69, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 
-----------descriptors: [{"uuid": "2902", "value": null}]
------------ [{"characteristic": "2a05", "properties": {"Indicate": "Indicate"}, "service": "1801"},
              {"characteristic": "22a4e311-a097-4517-9b81-cf32af60b982", descriptors: [{"uuid": "2902", "value": null}], 
             "properties": {"Indicate": "Indicate", "Notify": "Notify", "Write": "Write"},
              "service": "4553867f-f809-49f4-aefc-e190a1f459f3"}]

---FYI after decoding the data above we are able to see only the device name. Any idea

JulioCVaz, 2022-01-07:
@taj567 I'm fine :]

Congratz to connected with ble-manager and read data 👍 

I think that about your question, you can try convert this array buffer to a string or valid value.

```javascript
import { Buffer } from 'buffer'

const data = DATA_FROM_BLE_MANAGER

const bufferToString = Buffer.from(data.ArrayBuffer).toString('base64')

```
I see that this library is necessary to convert the received data buffer

https://github.com/innoveit/react-native-ble-manager#readperipheralid-serviceuuid-characteristicuuid

```javascript

BleManager.read(
  "XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX",
  "XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX",
  "XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX"
)
  .then((readData) => {
    // Success code
    console.log("Read: " + readData);

    const buffer = Buffer.Buffer.from(readData); //https://github.com/feross/buffer#convert-arraybuffer-to-buffer
    const sensorData = buffer.readUInt8(1, true);
  })
  .catch((error) => {
    // Failure code
    console.log(error);
  });

```

or follow this thread in the stackoverflow to create a buffer converter: https://stackoverflow.com/a/37902334/10608968

taj567, 2022-01-10:
@JulioCVaz Hi, 
if i convert the data using base64 decode input("AgEACwlQUiBCVCBGQzZGB/8AAAAAAAARB/NZ9KGQ4fyu9EkJ+H+GU0UAAAAAAAAAAAAAAAAAAAAAAAAAAAA=") = output(	PR BT FC6FYI	SE)--i can see the device name only

taj567, 2022-01-10:
@JulioCVaz by using this --BleManager.read(
  "XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX",
  "XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX",
  "XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX"
)
  .then((readData) => {
    // Success code
    console.log("Read: " + readData);

    const buffer = Buffer.Buffer.from(readData); //https://github.com/feross/buffer#convert-arraybuffer-to-buffer
    const sensorData = buffer.readUInt8(1, true);
  })
  .catch((error) => {
    // Failure code
    console.log(error);
  });

---getting read failed error---

JulioCVaz, 2022-01-10:
@taj567 Alright, I think that to read the data from your device, maybe u can use another library more robust like https://github.com/dotintent/react-native-ble-plx

I search the specification about your machine https://www.cpapdirect.com/cpap-machines/philips-respironics-dreamstation-auto-cpap-with-heated-humidifier but I don't find if is a machine with BLE(Bluetooth low energy) or a Normal Bluetooth connect. This is diferent to read of data.

taj567, 2022-01-11:
@JulioCVaz Yes, it is an updated one!.
Role: with Bluetooth pairing, we need to manage the device like turn it off & on or increasing the pressure
(ex: atmospere-connect).

taj567, 2022-01-11:
@JulioCVaz we are able to pair the device and connect but we need to get the extra information related to the device.

JulioCVaz, 2022-01-11:
@taj567 awesome! Try to use the `react-native-ble-plx` if u have a problem what do u think that we schedule a call?

dwsMeet, 2024-03-06:
Hi, @JulioCVaz How are you?

Can I retrieve data from the digital hemoglobin meter poc-30 point of care device via Bluetooth using react-native-bluetooth-serial, or do I need to use Android code?