rusel1989/react-native-bluetooth-serial #113: Who can show me a completely runnable case？

GHOSTLORE, 2019-12-05:
I want to learn how to use this plugin，but the document and demo is not friendly,wanna help!

caiorios, 2019-12-25:
Hello, I've created a file with custom hooks to help me using it. Hope it helps you!

```
import { useEffect, useCallback } from 'react';
import BluetoothSerial from 'react-native-bluetooth-serial';

const BLUETOOTH_ENABLED = 'bluetoothEnabled';
const BLUETOOTH_DISABLED = 'bluetoothDisabled';
const BLUETOOTH_ERROR = 'error';
const BLUETOOTH_CONNECTION_LOST = 'connectionLost';

export const useBluetooth = () => {
  const list = useCallback(async () => {
    const devices = await BluetoothSerial.list();
    console.log({ list: devices });
    return devices;
  }, []);

  const connect = useCallback(async deviceId => {
    const device = await BluetoothSerial.connect(deviceId);
    console.log({ connect: device });
    return device;
  }, []);

  const pairDevice = useCallback(async deviceId => {
    const isEnabled = await BluetoothSerial.enable();
    console.log({ isEnabled });
    const device = isEnabled
      ? await BluetoothSerial.pairDevice(deviceId)
      : null;
    console.log({ pairDevice: device });
    return device;
  }, []);

  const unpairDevice = useCallback(async id => {
    const isDisconnected = await BluetoothSerial.unpairDevice(id);
    console.log({ isDisconnected });
    return true;
  }, []);

  return { list, connect, unpairDevice, pairDevice };
};

export const useBluetoothEnabled = (callback, deps) => {
  useEffect(() => {
    BluetoothSerial.on(BLUETOOTH_ENABLED, callback);

    return () => {
      BluetoothSerial.removeListener(BLUETOOTH_ENABLED, callback);
    };
    // eslint-disable-next-line react-hooks/exhaustive-deps
  }, deps);
};

export const useBluetoothDisabled = (callback, deps) => {
  useEffect(() => {
    BluetoothSerial.on(BLUETOOTH_DISABLED, callback);

    return () => {
      BluetoothSerial.removeListener(BLUETOOTH_DISABLED, callback);
    };
    // eslint-disable-next-line react-hooks/exhaustive-deps
  }, deps);
};

export const useBluetoothError = (callback, deps) => {
  useEffect(() => {
    BluetoothSerial.on(BLUETOOTH_ERROR, callback);

    return () => {
      BluetoothSerial.removeListener(BLUETOOTH_ERROR, callback);
    };
    // eslint-disable-next-line react-hooks/exhaustive-deps
  }, deps);
};

export const useBluetoothConnectionLost = (callback, deps) => {
  useEffect(() => {
    BluetoothSerial.on(BLUETOOTH_CONNECTION_LOST, callback);

    return () => {
      BluetoothSerial.removeListener(BLUETOOTH_CONNECTION_LOST, callback);
    };
    // eslint-disable-next-line react-hooks/exhaustive-deps
  }, deps);
};
```

litesam, 2020-02-29:
@caiorios Can you please share the whole project, and share it here!?

litesam, 2020-03-01:
Hey, @GHOSTLORE a sample app that I made for College as a project, It's working fine you can view it   https://github.com/litesam/bluevoice.

Also, I haven't used hooks since my friend who is working with me on the project haven't learned hooks so I was using the old way of writing components as classes.

caiorios, 2020-03-04:
Hello @litesam. I end up forking this project and made some minors changes. I didn't add hooks helpers in this project yet. If this interest you, here is the project:

https://github.com/caiorios/rn-bluetooth

litesam, 2020-03-04:
Hey, @caiorios. I saw your project looks it is cool since I'm new to React Native I'm just writing things on the go, will have to refactor after finishing the project prototype.

amrudesh-santhanam, 2020-03-05:
Hi @caiorios . I tried https://github.com/caiorios/rn-bluetooth too. I am able to get the app working but it is not detecting unpaired devices while other apps are able to see the devices. I had the same problem with this repository. I am trying on android. I see tried both methods; by doing react-native link and also by the changes to MainApplication.java, android/settings.gradle and android/app/build.gradle. Both give me same problem (unable to discover devices). 

For context, I am doing a react-native init BluetoothSerialExample and then copying in the code from your repository. 

What else should I try? Am i missing something.