rusel1989/react-native-bluetooth-serial #115: :react-native-bluetooth-serial:compileDebugJavaWithJavac

fayistuh, 2019-12-14:
any idea??

yatagan, 2019-12-28:
@fayistuh try this https://github.com/react-native-community/react-native-device-info/issues/213#issuecomment-338479090

fayistuh, 2019-12-30:
@yatagan But where is RNDeviceInfo.java file is existing

fayistuh, 2019-12-30:
* What went wrong:
Execution failed for task ':react-native-bluetooth-serial:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

AristideYao, 2020-04-06:
Have a look to this:
https://github.com/rusel1989/react-native-bluetooth-serial/issues/97

mateus4k, 2020-05-13:
> Have a look to this: #97
> It now works fo since I have:
> completely remove @override at line 23 of files : RCTBluetoothSerialPackage
> inside the package: node_modules\react-native-bluetooth-serial\android\src\main\java\com\rusel\RCTBluetoothSerial

Wow! It's works here... but until I send it to remote repository and lost the "custom node_modules" folder... 😅

rudved, 2021-04-24:
Task:react-native-bluetooth-serial:compileDebugJavaWithJavac FAILED
getting this error if anyone knw help me

YashKalavadiya, 2022-02-05:
Same problem. Works when removed @override in node_modules. Need more permanent solution.!!

ambojialur, 2022-09-21:
> Have a look to this: #97 It now works fo since I have: completely remove @OverRide at line 23 of files : RCTBluetoothSerialPackage inside the package: node_modules\react-native-bluetooth-serial\android\src\main\java\com\rusel\RCTBluetoothSerial

Thank you