rusel1989/react-native-bluetooth-serial #134: Data read from arduino HC05 is always empty

damodharanj, 2023-08-30:
`BluetoothSerial.withDelimiter('\r\n').then(() => {
          setInterval(() => {
            // BluetoothSerial.readUntilDelimiter("\n").then((v: any) => {
            //   setVal(JSON.stringify(v));
            // })
            BluetoothSerial.available().then((a: any) => {
                setVal(a);
            })
          }, 3000);
        })`

The data i read from arduino HC05 is always empty. But bluetooth terminal app shows correct text

CS-Usman, 2023-11-22:
Did your problem got solved ? I am also working on HC-05 module need your help

damodharanj, 2023-11-27:
You gotta connect to the device before you read. Its a clutch unless you see the example code.